JuliaRobotics/RigidBodyDynamics.jl #58: Duplicate body names when attaching two mechanisms

rdeits, 2016-10-06:
I have been (implicitly) assuming in RigidBodyTreeInspector that all the links in a robot will have unique names. That assumption comes from the `drake-visualizer` interface, which uses the link names as the only way of identifying links within a robot. However, when I `attach!()` one mechanism onto another, the combined mechanism will have duplicate link names. 

How do you want to handle this situation? Should `RigidBodyDynamics` enforce uniqueness of the link names? Or should I just make `RigidBodyTreeInspector` more robust to this case? Really, I should do the latter anyway, but it's still good to decide if uniqueness of link names is a safe assumption.

tkoolen, 2016-10-06:
I don't want to rename things by default, but I can make `RigidBody` and `Joint` `type`s instead of `immutable`s (that's necessary to implement efficient parameter optimization anyway) and add a `make_names_unique!` utility function that renames the bodies and joints. Any preference for how exactly to rename things?

rdeits, 2016-10-07:
Ok, cool. I don't have any opinions about how to rename, except that I think it probably makes more sense to change the names of the new bodies being attached, rather than changing the names of the existing bodies.

rdeits, 2016-10-07:
We decided offline that changing the body names here isn't worth the extra effort. I will just not assume that all bodies have unique names, and will update RigidBodyTreeInspector to work anyway.

closed 2016-10-07 by rdeits