JuliaRobotics/RigidBodyDynamics.jl #63: Error when constructing MechanismState for an attached mechanism

rdeits, 2016-10-07:
When constructing a new `MechanismState` after attaching two mechanisms, I get one of two possible errors:

```julia
julia> MechanismState(Float64, mechanism)
ERROR: KeyError: key CartesianFrame3D: "world" not found
 in getindex at ./dict.jl:688 [inlined]
 in RigidBodyDynamics.MechanismState{X<:Real,M<:Real,C<:Real}(::Type{Float64}, ::RigidBodyDynamics.Mechanism{Float64}) at /Users/rdeits/locomotion/explorations/point-cloud-signed-distance/packages/v0.5/RigidBodyDynamics/src/mechanism_state.jl:196
```

or 

```julia
julia> MechanismState(Float64, mechanism)
ERROR:
 in getindex at ./dict.jl:688 [inlined]
 in RigidBodyDynamics.MechanismState{X<:Real,M<:Real,C<:Real}(::Type{Float64}, ::RigidBodyDynamics.Mechanism{Float64}) at /Users/rdeits/locomotion/explorations/point-cloud-signed-distance/packages/v0.5/RigidBodyDynamics/src/mechanism_state.jl:196SYSTEM: show(lasterr) caused an error
```

Which error I get is unpredictable. A given `mechanism` instance will always throw the same error every time I try to construct `MechanismState`, but running the exact same code (with the RNG seed fixed) to generate a new `mechanism` may result in one which throws the other error. 

The code I'm using to generate the attached mechanism is:

```julia
srand(0)
mechanism = rand_tree_mechanism(Float64, [Revolute{Float64} for i = 1 : 2]...)
mechanism2 = rand_tree_mechanism(Float64, [Revolute{Float64} for i = 1 : 2]...)
connection = Joint("connection", rand(Revolute{Float64}))
parentBody = collect(bodies(mechanism))[2]
attach!(mechanism, parentBody, connection, rand(Transform3D{Float64}, connection.frameBefore, parentBody.frame), mechanism2)
```

rdeits, 2016-10-07:
It looks like the frame IDs are getting mangled:

```julia
julia> for v in mechanism.toposortedTree
           @show v.vertexData.inertia.frame.id
       end
v.vertexData.inertia.frame.id = 0
v.vertexData.inertia.frame.id = 3
v.vertexData.inertia.frame.id = 7
v.vertexData.inertia.frame.id = 569348
v.vertexData.inertia.frame.id = 12
v.vertexData.inertia.frame.id = 16

julia> keys(RigidBodyDynamics.frame_names)
Base.KeyIterator for a Dict{Int64,String} with 20 entries. Keys:
  18
  2
  16
  11
  0
  7
  9
  10
  19
  17
  8
  6
  4
  3
  5
  13
  14
  15
  12
  1
```

tkoolen, 2016-10-08:
Frame IDs getting mangled: that's for the root body. The `RigidBody` constructor that only takes a `String` is broken; it's not setting the `inertia` member and I was thinking that would make it `!isdefined`, but that's not possible for `immutable` members (kind of a gotcha in the language in my opinion). I'll fix this next.

commit e4d1af41 (2016-10-08): Completely revamp attach!ing another mechanism.

Previously, it was conceptually unclear what it meant to attach! mechanism2 to a body of mechanism1, since we were attaching the root body (world) of mechanism2 to mechanism1, and the inertia of mechanism2's root body is undefined. The new interpretation of attach!ing mechanism2 to parentBody (part of mechanism1) is replacing the root body of mechanism2 with parentBody. Note that mechanism2 should be unchanged after attach!ing it to mechanism1. This change also cleans up the function signature.

Fixes #63.

Also made RigidBody and Joint types in [... cut]
PR #66 (2016-10-08): Improve attach!
Previously, it was conceptually unclear what it meant to attach! mechanism2 to a body of mechanism1, since we were attaching the root body (world) of mechanism2 to mechanism1, and the inertia of mechanism2's root body is undefined. The new interpretation of attach!ing mechanism2 to parentBody (part of mechanism1) is replacing the root body of mechanism2 with parentBody. Note that mechanism2 should be unchanged after attach!ing it to mechanism1. This change also cleans up the function signature.

Fixes #63.

Also made RigidBody and Joint types instead of immutables, among other  [... cut]
closed 2016-10-08 by tkoolen