JuliaRobotics/RigidBodyDynamics.jl #79: Crazy idea: static char arrays instead of global frame_names

rdeits, 2016-10-14:
The global `frame_names` map has a somewhat scary problem in that every new coordinate frame causes that list to grow, and old entries can never be deleted. So far this hasn't actually cause any measurable issues for me, but it's still concerning. 

What if, instead, we used a fixed-size static array of Chars instead of the Int64 reference? That might look something like:

```julia
immutable CartesianFrame3D
    name::SVector{64, Char}
    function CartesianFrame3D(name::String)
        length(name) > 64 && warn("Frame name longer than 64 characters was truncated")
        new(SVector{64, Char}(Base.flatten((name, ('\x0' for i in 1:(64 - length(name)))))...))
    end
end
    
show(io::IO, frame::CartesianFrame3D) = print(io, "CartesianFrame3D: $(join(frame.name[frame.name .!= '\x0'], ""))")
```

This has the obvious problem that it makes the stack-allocated size of a frame much, much larger. That alone might be enough to make this a very bad idea. However, it does result in an `isbits` type that doesn't require a global reference.

tkoolen, 2016-10-16:
I see this as a place where the language is currently a little bit lacking. My plan was to stick with the current hack until Julia improves. Related Julia issues/PRs:
* https://github.com/JuliaLang/julia/pull/8134
* https://github.com/JuliaLang/julia/pull/12205 and https://groups.google.com/d/msg/julia-users/LthfABeDN50/kKBgB-WQAAAJ
* https://github.com/JuliaLang/julia/issues/14955
* https://github.com/JuliaLang/julia/issues/16190
* https://github.com/JuliaLang/julia/issues/10899 (part of 1.0 milestone)

If this becomes a real problem for you or anybody else, I can try what you proposed. I guess for now, try not to create new frames at every time step of your control algorithm or simulation, or if you must, manually remove the `frame_names` entries for frames that will go out of scope. Unfortunately I can't use `finalizer` either since "the behavior of this function is unpredictable if x is of a bits type".

rdeits, 2016-10-16:
Ok, makes sense. I'll let you know if the hack ever causes real problems.

tkoolen, 2016-10-29:
Another option is to make it possible to have anonymous frames for transient use (e.g. have `id == -1` mean anonymous frame, and have `name` return "(anonymous frame)" in this case).

tkoolen, 2017-06-29:
You know what, I'm pretty happy with the current solution; no issues related to this after months, so I'll close this.

closed 2017-06-29 by tkoolen