JuliaRobotics/RigidBodyDynamics.jl #83: Contact dynamics

tkoolen, 2016-10-22:
- [ ] Store collision geometry in `Mechanism` (use GeometryTypes?).
- [ ] Decide how to do distance queries (GJK implementation from GeometryTypes?).
- [ ] Incorporate contact in dynamics. Currently leaning towards an LCP-style implementation.
- [ ] Parsing collision elements from URDF.

rdeits, 2016-10-22:
Having geometry stored inside the mechanism would make RigidBodyTreeInspector's job much easier 👍

rdeits, 2016-10-22:
Also, the GJK implementation from GeometryTypes will need to be improved before it can be useful here. It's got two important features missing: 

* it repeatedly searches over all of the nodes in the mech to find the closest points, which is *extremely* inefficient. Better implementations use the mesh edges to only look at adjacent points in the mesh.
* it doesn't compute the penetration distance when the objects are in collision

Both of these are on my radar for my research too, since GJK is the limiting step in my tracking code.

beel7217, 2020-02-27:
Where are the current contact dynamics implemented? The only thing I can tell from the documentation is that contact points can be added to a RigidBody, but it is very unclear how the contact or 'ground' plane is defined and accessed. Is the computation of contact forces implicit when using dynamics! with a floating base model? Any clarification would be greatly appreciated.