JuliaRobotics/RigidBodyDynamics.jl #87: Consider switching from Quaternions.jl to Rotations.jl or CoordinateTransformations.jl

tkoolen, 2016-10-22:
See https://github.com/tkoolen/RigidBodyDynamics.jl/commit/866fca5354c228cd15302176bbbd3cab500ec433.

Quaternions.jl is barely maintained, and not having to implement conversions to other rotation parameterizations in RigidBodyDynamics.jl would be nice.

Cons of switching to Rotations.jl: currently, Rotations.jl's quaternion type is really a unit quaternion only, and always normalizes upon construction. There are a couple of places where it would be nice to have non-normalized quaternions (mostly for quaternion derivatives). Also, it wouldn't be possible to get the dynamics in trigonometric polynomial form if the quaternion is normalized upon construction.

andyferris, 2016-10-27:
Points well taken. In *Rotations.jl*, we took the decision to remove the mental load of keeping track of normalization on the user.

> There are a couple of places where it would be nice to have non-normalized quaternions (mostly for quaternion derivatives)

There are some derivatives already implemented in *Rotations.jl*. We are also thinking hard about good interfaces for derivatives in *CoordinateTransformations.jl*, but it isn't really there yet.

> Also, it wouldn't be possible to get the dynamics in trigonometric polynomial form if the quaternion is normalized upon construction.

You could try expansions which are compositions of valid rotations. These often have some desirable properties (e.g. can make a symplectic integrator quite easily).

c42f, 2016-10-27:
> dynamics in trigonometric polynomial form

@tkoolen perhaps you can expand a bit on the formulation you're trying to set up? I'd be really happy to collaborate on making Rotations and CoordinateTransformations fit for use in dynamics and with kinematic chains. This is something @PaulBellette and I have discussed a few times, but it's not our current use case so we haven't pursued it.

Generally speaking, we currently aim for CoordinateTransformations to be a great package for optimization, which includes all kinds of derivatives and should also include dealing with constraints on the transformation parameterizations.  I'm really keen to learn from your use case and make the package great for everyone.

tkoolen, 2016-10-27:
Renormalization is good for numeric types like `Float64`, but maybe not for some other scalar types. 

There may be cases where the equations of motion are required in symbolic form. For a symbolic scalar type (e.g. from SymPy.jl), renormalizing for numerical reasons isn't necessary and complicates the resulting expression. In the symbolic case, perhaps you can simplify the expression after the fact, but it gets a little more annoying for trigonometric polynomials. 

In Drake (a toolbox made in my lab), there is a [`TrigPoly`](https://github.com/RobotLocomotion/drake/blob/5802d91eb590a6e63d9a28f3b01ac48b56353656/drake/matlab/util/TrigPoly.m) scalar type which would be nice to have in Julia eventually. It does the following: whenever `sin` or `cos` is called on a `TrigPoly` `x_i`, two variables, `s_i = sin(x_i)` and `c_i = cos(x_i)`, are created or looked up from a table if they already exist and substituted into the expression. In addition, the constraint `s_i^2 + c_i^2 == 1` is added to a list of constraints. What this does is despite the fact that e.g. the mass matrix is a trigonometric function of the joint positions, the resulting expression will be a *polynomial* function of the `s_i`, `c_i` (and perhaps some of the `x_i`) variables + a list of polynomial constraints on the `s_i`, `c_i` variables. This makes the dynamics of a general mechanism amenable to techniques like region of attraction analysis using sum of squares optimization (see e.g. https://github.com/rdeits/Mayday.jl) or using SMT solvers (see e.g. https://github.com/dreal/DReal.jl). The issue here is that you don't really want to define division and square root for the trigonometric polynomial scalar type, because the result of those operations is no longer polynomial, so it doesn't play so nice with renormalization.

Having typed all this up, I now realize that maybe this case will also be fine with renormalization if you do simplifications of the form `a * s_i^2 + a * c_i^2 --> a` for constant `a` while evaluating the expressions (with a numerical epsilon used to check that the coefficients of `s_i^2` and `c_i^2` are approximately the same). I still think it's cleaner not to have to worry about renormalization though.

I wonder if a trait specifying whether to do renormalization upon Quaternion construction would be a good option.

tkoolen, 2016-10-27:
> trait specifying whether to do renormalization

Or just a function such as `Rotations.perform_quaternion_normalization{T}(::Type{T})` that returns a boolean and can be overloaded by users.

tkoolen, 2016-12-04:
OK, decided I will switch soon.

I did some basic benchmarking and found that my assumption of quaternions being faster than rotation matrices was wrong (probably due to better use of SIMD instructions?). So I'll switch to rotation matrices as the main workhorse used in `Transform3D`. This means my concern over automatic renormalization is alleviated somewhat (at least my performance concern, which I was worried about but hadn't mentioned before).

I'll still need to use quaternions in `QuaternionFloating` joints however. Since I'm not currently working on the symbolic stuff I mentioned in https://github.com/tkoolen/RigidBodyDynamics.jl/issues/87#issuecomment-256708465, I'm going to not worry about renormalization for now. I could always switch the floating joint type to using a rotation matrix as well at a later stage.

I will say that I still think automatic renormalization of quaternions is an unfortunate design choice. It's also not consistent with rotation matrices, where no such automatic renormalization is performed (and would be quite expensive, please don't change `RotMatrix` to automatically renormalize as well :-) ).

closed 2016-12-06 by tkoolen