JuliaRobotics/RigidBodyDynamics.jl #170: `show()` only shows the root link of a mechanism

rdeits, 2017-02-15:
It used to show the whole tree. Was that an intentional change? 

```julia
julia> mechanism = rand_chain_mechanism(Float64, [QuaternionFloating{Float64}; [Revolute{Float64} for i = 1:5]]...)
Vertex: world (root)

julia> tree(mechanism)
Vertex: world (root)
  Vertex: body1, Edge: joint1
    Vertex: body2, Edge: joint2
      Vertex: body3, Edge: joint3
        Vertex: body4, Edge: joint4
          Vertex: body5, Edge: joint5
            Vertex: body6, Edge: joint6
```

rdeits, 2017-02-15:
cc @symplectomorphism

tkoolen, 2017-02-16:
Thanks. I was trying to get `show` for an array of TreeVertices to work correctly (and not print the whole subtree for each element of the array), but I messed it up.

commit ac923e53 (2017-02-16): Merge pull request #173 from tkoolen/fix-170

Fix #170
commit b06aafb8 (2017-02-16): Fix #170.
closed 2017-02-16 by tkoolen