JuliaRobotics/RigidBodyDynamics.jl #184: Generalizing the diffeqs

ChrisRackauckas, 2017-03-17:
Hey,
  I was wondering what kinds of diffeq methods you guys would like to see, as well as what a good general API would be. I see that you use Lie group integrators here, and I am not too familiar with them, so I was wondering if you could help JuliaDiffEq design an interface for Lie Group integrators that methods can be built off of. Specifically:

- What would the data of a `LieGroupProblem` need to have? It would have the ODE `f` of course, but how would the Lie group be specified? Is there a good general way to do this?
- How do these compare to other symplectic methods?
- Do you have an internal version of `expmv` you're using for this?

tkoolen, 2017-03-21:
Oops, sorry I missed this issue.

> I see that you use Lie group integrators here, and I am not too familiar with them, so I was wondering if you could help JuliaDiffEq design an interface for Lie Group integrators that methods can be built off of

Well, just the one, `MuntheKaasIntegrator`. Yeah, sure, I'd like to help. Disclaimer: I'm not really an expert on integrators. I did however recently give a short presentation about Lie group integrators at our lab's group meeting: [lie-group-methods.pdf](https://github.com/tkoolen/RigidBodyDynamics.jl/files/857175/lie-group-methods.pdf).

> What would the data of a `LieGroupProblem` need to have? It would have the ODE `f` of course, but how would the Lie group be specified? Is there a good general way to do this?

Just a note about `f` first: the integrator I made is specifically for second-order ODEs, so in my case `f` only returns accelerations (time derivatives of tangent vectors). In fact, the integrator doesn't require computation of the time derivatives of the configurations at all. There are obviously also use cases for first-order ODEs, but this is something to keep in mind.

Sticking to first-order ODEs for now, `f` should take a group element and return a tangent vector, so in general f(t, x) ≠ ẋ, and the dimension of x may not be the same as that of f(t, x).

As for how the group should be specified, I think it depends on the method you use. The two main methods that I know of are Crouch-Grossman and Munthe-Kaas (see the presentation I attached and references therein):
* With a Crouch-Grossman method, the stages are group elements. For each stage, you compute the tangent vector using `f`, scale that tangent vector by an integrator coefficient (an a_{i, j} in a Butcher tableau), and convert the scaled tangent vector to a group element using the exponential map. You then compose the group elements for the stages and the group element at the start of the integration step using the group multiplication operation to obtain the next group element at which to evaluate `f`. Combining the stages into the final step is also done using the group multiplication operation. Downsides of this approach are that you need a lot of expensive exponential map evaluations and that you need to use a very specific Butcher tableau to satisfy the order conditions (you can't just use e.g. RK4 coefficients), and the only 4th order Crouch-Grossman method requires 5 stages.
* With a Munthe-Kaas method, the stages are local coordinates (often, but not necessarily, exponential coordinates) around the group element at the start of the integration step. Again, you compute a tangent vector at each stage using `f`, but now you don't convert the tangent vector to a group element. Instead, you convert the tangent vector to time derivatives of local coordinates, and then you combine these local coordinate derivative vectors using scaling and addition as you would with a normal RK4 method (you can use any 'normal' integrator coefficients for this). However, at each stage you do need to convert the local coordinates around the initial group element to a new group element, so that you can evaluate `f` at that new group element.

So for a Crouch-Grossman method, you need the user to supply the exponential map (Lie algebra element to group element) and the group multiplication operation. For a Munthe-Kaas method, you need a way to go from local coordinates around a group element to a new group element (possibly using the same exponential map), as well as a way to go from a tangent vector to a vector of local coordinate derivatives.

> How do these compare to other symplectic methods?

Again, not an expert, but I believe symplectic methods are orthogonal to Lie group methods. The Munthe-Kaas integrator I implemented is explicit, and I think you'd need an implicit integrator for a symplectic method. See the Park and Chung reference in my slides for some comments about that.

> Do you have an internal version of `expmv` you're using for this?

How to implement the exponential map / conversions between local coordinates and group elements will in general depend on the particular group. For example, for rotations parameterized using a rotation matrix, you could use the generic `expm` on the skew symmetric matrix associated with the angular velocity vector, but using [Rodrigues' rotation formula](https://en.wikipedia.org/wiki/Rodrigues'_rotation_formula) will be much faster and likely more accurate. For quaternions, the exponential map is defined completely differently.

The implementation I use is for conversion from a twist (a 6D vector that describes both angular and linear motion) to a homogeneous transform; see here: https://github.com/tkoolen/RigidBodyDynamics.jl/blob/dfbf7aac31875c69ea8def33400baaa387822bbb/src/spatial.jl#L436.

ChrisRackauckas, 2017-03-21:
Thanks! I'll add this to the list!

closed 2017-04-11 by tkoolen