JuliaRobotics/RigidBodyDynamics.jl #188: Update UnsafeVectorView (doesn't work for non-isbits types)

tkoolen, 2017-03-28:
Based on @rdeits's changes in https://github.com/rdeits/NNLS.jl/blob/0a9bf56774595b5735bc738723bd3cb94138c5bd/src/NNLS.jl#L200.

rdeits, 2017-04-11:
I was trying it with BigFloat, which doesn't work: 

```julia
julia> x = zeros(BigFloat, 5);

julia> UnsafeVectorView(x, 1, 1)
1-element UnsafeVectorView{BigFloat}:
Error showing value of type UnsafeVectorView{BigFloat}:
ERROR: InexactError()
 in trunc(::Type{Int32}, ::Float64) at ./float.jl:458
 in string(::BigFloat) at ./mpfr.jl:853
 in show(::IOContext{Base.AbstractIOBuffer{Array{UInt8,1}}}, ::BigFloat) at ./mpfr.jl:869
 in #sprint#316(::IOContext{Base.Terminals.TTYTerminal}, ::Function, ::Int64, ::Function, ::BigFloat, ::Vararg{BigFloat,N}) at ./strings/io.jl:35
 in (::Base.#kw##sprint)(::Array{Any,1}, ::Base.#sprint, ::Int64, ::Function, ::BigFloat, ::Vararg{BigFloat,N}) at ./<missing>:0
 in alignment(::IOContext{Base.Terminals.TTYTerminal}, ::BigFloat) at ./show.jl:1240
 in alignment(::IOContext{Base.Terminals.TTYTerminal}, ::UnsafeVectorView{BigFloat}, ::Array{Int64,1}, ::Array{Int64,1}, ::Int64, ::Int64, ::Int64) at ./show.jl:1278
 in print_matrix(::IOContext{Base.Terminals.TTYTerminal}, ::UnsafeVectorView{BigFloat}, ::String, ::String, ::String, ::String, ::String, ::String, ::Int64, ::Int64) at ./show.jl:1407
 in print_matrix(::IOContext{Base.Terminals.TTYTerminal}, ::UnsafeVectorView{BigFloat}, ::String, ::String, ::String) at ./show.jl:1379
 in #showarray#342(::Bool, ::Function, ::IOContext{Base.Terminals.TTYTerminal}, ::UnsafeVectorView{BigFloat}, ::Bool) at ./show.jl:1618
 in display(::Base.REPL.REPLDisplay{Base.REPL.LineEditREPL}, ::MIME{Symbol("text/plain")}, ::UnsafeVectorView{BigFloat}) at ./REPL.jl:132
 in display(::Base.REPL.REPLDisplay{Base.REPL.LineEditREPL}, ::UnsafeVectorView{BigFloat}) at ./REPL.jl:135
 in display(::UnsafeVectorView{BigFloat}) at ./multimedia.jl:143
 in print_response(::Base.Terminals.TTYTerminal, ::Any, ::Void, ::Bool, ::Bool, ::Void) at ./REPL.jl:154
 in print_response(::Base.REPL.LineEditREPL, ::Any, ::Void, ::Bool, ::Bool) at ./REPL.jl:139
 in (::Base.REPL.##22#23{Bool,Base.REPL.##33#42{Base.REPL.LineEditREPL,Base.REPL.REPLHistoryProvider},Base.REPL.LineEditREPL,Base.LineEdit.Prompt})(::Base.LineEdit.MIState, ::Base.AbstractIOBuffer{Array{UInt8,1}}, ::Bool) at ./REPL.jl:652
 in run_interface(::Base.Terminals.TTYTerminal, ::Base.LineEdit.ModalInterface) at ./LineEdit.jl:1579
 in run_frontend(::Base.REPL.LineEditREPL, ::Base.REPL.REPLBackendRef) at ./REPL.jl:903
 in run_repl(::Base.REPL.LineEditREPL, ::Base.##950#951) at ./REPL.jl:188
 in _start() at ./client.jl:363
```

(the error happens inside `show()` because the value stored in the BigFloat becomes some crazy-large random number). 

Maybe something about the memory layout of Ref lets you avoid the issue for that particular type?

tkoolen, 2017-04-11:
OK, yep, you're right, and the Ref example also turned out to be bad.

tkoolen, 2017-04-11:
You might want to be a bit more conservative in NNLS.jl with the type of array you allow as the parent BTW. `DenseArray` doesn't necessarily imply that the data will be laid out contiguously in memory if an isbits type is used. I'm going to go with `Union{Vector, Base.FastContiguousSubArray}` in RigidBodyDynamics.

closed 2017-04-11 by tkoolen

rdeits, 2017-04-11:
Ah, good point. I didn't know that.

PR #206 (2017-04-11): Use fastview, use to select between UnsafeVectorView and SubArray.
Slightly tweaked from NNLS.jl 
Fixes #188.
commit b2dc1995 (2017-04-11): Add fastview, fix #188.