JuliaRobotics/RigidBodyDynamics.jl #221: [v0.6-rc2] Error running the symbolic double pendulum example

smldis, 2017-05-28:
I get this error from both release and master branch while running the example at: <http://nbviewer.jupyter.org/github/tkoolen/RigidBodyDynamics.jl/blob/master/notebooks/Symbolic%20double%20pendulum.ipynb>

```
MethodError: convert(::Type{RigidBodyDynamics.SpatialInertia{SymPy.Sym}}, ::RigidBodyDynamics.SpatialInertia{SymPy.Sym}) is ambiguous. Candidates:
  convert(::Type{RigidBodyDynamics.SpatialInertia{T}}, inertia::RigidBodyDynamics.SpatialInertia) where T<:Number in RigidBodyDynamics at /home/smldis/.julia/v0.6/RigidBodyDynamics/src/spatial.jl:185
  convert(::Type{RigidBodyDynamics.SpatialInertia{T}}, inertia::RigidBodyDynamics.SpatialInertia{T}) where T in RigidBodyDynamics at /home/smldis/.julia/v0.6/RigidBodyDynamics/src/spatial.jl:182
Possible fix, define
  convert(::Type{RigidBodyDynamics.SpatialInertia{T<:Number}}, ::RigidBodyDynamics.SpatialInertia{T<:Number})

Stacktrace:
 [1] Type at /home/smldis/.julia/v0.6/RigidBodyDynamics/src/rigid_body.jl:24 [inlined]
 [2] RigidBodyDynamics.RigidBody(::RigidBodyDynamics.SpatialInertia{SymPy.Sym}) at /home/smldis/.julia/v0.6/RigidBodyDynamics/src/rigid_body.jl:31
 [3] execute_request(::ZMQ.Socket, ::IJulia.Msg) at /home/smldis/.julia/v0.6/IJulia/src/execute_request.jl:156
 [4] eventloop(::ZMQ.Socket) at /home/smldis/.julia/v0.6/IJulia/src/eventloop.jl:8
 [5] (::IJulia.##9#12)() at ./task.jl:335
```

For now I hacked it away by deleting the line 182 of spatial.jl.

Do you get the same error?

tkoolen, 2017-06-05:
Thanks for reporting and sorry about the response time. I've been traveling for the last two weeks. I'll look at this today.

tkoolen, 2017-06-05:
Hmm, I can't actually reproduce this on master with RC2. Are you using the latest RigidBodyDynamics master?

smldis, 2017-06-05:
Ah nice, with latest master it works.
Closing!

closed 2017-06-05 by smldis