JuliaRobotics/RigidBodyDynamics.jl #330: ForwardDiff v0.5.0 hessians not working through RBD functions

goretkin, 2017-09-10:
I haven't been able to narrow down what the problem is, so this issue is quite verbose.

I'm trying to take ForwardDiff.hessians of RigidBodyDynamics functions: https://gist.github.com/goretkin/e858363daa6a4740934f4fbc48020edc#file-hessians_rbd-jl-L109 and I am getting 

```
MethodError: Cannot `convert` an object of type RigidBodyDynamics.CustomCollections.UnsafeFastDict{RigidBodyDynamics.Graphs.vertex_index,RigidBodyDynamics.RigidBody{Float64},Array{Array{RigidBodyDynamics.Contact.SoftContactStateDeriv{Void,RigidBodyDynamics.Contact.ViscoelasticCoulombStateDeriv{SubArray{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{_,Float64,2},2},1},Tuple{UnitRange{Int64}},true}}},1},1} where _<:ForwardDiff.Tag} to an object of type RigidBodyDynamics.CustomCollections.UnsafeFastDict{RigidBodyDynamics.Graphs.vertex_index,RigidBodyDynamics.RigidBody{Float64},Array{Array{RigidBodyDynamics.Contact.SoftContactStateDeriv{Void,RigidBodyDynamics.Contact.ViscoelasticCoulombStateDeriv{SubArray{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1},Tuple{UnitRange{Int64}},true}}},1},1}}
This may have arisen from a call to the constructor RigidBodyDynamics.CustomCollections.UnsafeFastDict{RigidBodyDynamics.Graphs.vertex_index,RigidBodyDynamics.RigidBody{Float64},Array{Array{RigidBodyDynamics.Contact.SoftContactStateDeriv{Void,RigidBodyDynamics.Contact.ViscoelasticCoulombStateDeriv{SubArray{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1},Tuple{UnitRange{Int64}},true}}},1},1}}(...),
since type constructors fall back to convert methods.
Stacktrace:
 [1] RigidBodyDynamics.DynamicsResult{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},M} where M(::RigidBodyDynamics.Mechanism{Float64}) at /Users/goretkin/playgrounds/jump/packages/v0.6/RigidBodyDynamics/src/dynamics_result.jl:73
 [2] evaluate_friction_kinetic_energy(::RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}, ::Float64, ::Float64, ::Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1}) at /Users/goretkin/projects/julia_lqr/min_jump.jl:63
 [3] vector_mode_gradient at /Users/goretkin/playgrounds/jump/packages/v0.6/ForwardDiff/src/gradient.jl:94 [inlined]
 [4] gradient(::##2#3, ::Array{ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},1}, ::ForwardDiff.GradientConfig{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1}}) at /Users/goretkin/playgrounds/jump/packages/v0.6/ForwardDiff/src/gradient.jl:19
 [5] vector_mode_jacobian(::ForwardDiff.##43#44{##2#3,ForwardDiff.HessianConfig{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1},0xb046287d533b082d,Array{ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},1}}}, ::Array{Float64,1}, ::ForwardDiff.JacobianConfig{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},1}}) at /Users/goretkin/playgrounds/jump/packages/v0.6/ForwardDiff/src/jacobian.jl:132
 [6] jacobian(::ForwardDiff.##43#44{##2#3,ForwardDiff.HessianConfig{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{##2#3,0x096989b645fcecd8},ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},2},1},0xb046287d533b082d,Array{ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},1}}}, ::Array{Float64,1}, ::ForwardDiff.JacobianConfig{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{Void,0xb046287d533b082d},Float64,2},1}}) at /Users/goretkin/playgrounds/jump/packages/v0.6/ForwardDiff/src/jacobian.jl:21
 [7] hessian(::##2#3, ::Array{Float64,1}) at /Users/goretkin/playgrounds/jump/packages/v0.6/ForwardDiff/src/hessian.jl:19
 [8] include_from_node1(::String) at ./loading.jl:569
 [9] include(::String) at ./sysimg.jl:14
```

It's hard to read all those nested types, here they are indented:
from type:
```
RigidBodyDynamics.CustomCollections.UnsafeFastDict{
  RigidBodyDynamics.Graphs.vertex_index,
  RigidBodyDynamics.RigidBody{
    Float64},
  Array{
    Array{
      RigidBodyDynamics.Contact.SoftContactStateDeriv{
        Void,
        RigidBodyDynamics.Contact.ViscoelasticCoulombStateDeriv{
          SubArray{
            ForwardDiff.Dual{
              ForwardDiff.Tag{
                ##2#3,
                0x096989b645fcecd8},
              ForwardDiff.Dual{
                ForwardDiff.Tag{
                  Void,
                  0xb046287d533b082d},
                Float64,
                2},
              2},
            1,
            Array{
              ForwardDiff.Dual{
                ForwardDiff.Tag{
                  ##2#3,
                  0x096989b645fcecd8},
                ForwardDiff.Dual{
                  _,          # this line differs
                  Float64,
                  2},
                2},
              1},
            Tuple{
              UnitRange{
                Int64}},
            true}}},
      1},
    1}
where _<:ForwardDiff.Tag} # this line differs
```

to type:
```
RigidBodyDynamics.CustomCollections.UnsafeFastDict{
  RigidBodyDynamics.Graphs.vertex_index,
  RigidBodyDynamics.RigidBody{
    Float64},
  Array{
    Array{
      RigidBodyDynamics.Contact.SoftContactStateDeriv{
        Void,
        RigidBodyDynamics.Contact.ViscoelasticCoulombStateDeriv{
          SubArray{
            ForwardDiff.Dual{
              ForwardDiff.Tag{
                ##2#3,
                0x096989b645fcecd8},
              ForwardDiff.Dual{
                ForwardDiff.Tag{
                  Void,
                  0xb046287d533b082d},
                Float64,
                2},
              2},
            1,
            Array{
              ForwardDiff.Dual{
                ForwardDiff.Tag{
                  ##2#3,
                  0x096989b645fcecd8},
                ForwardDiff.Dual{
                  ForwardDiff.Tag{      # there is a concrete tag
                    Void,
                    0xb046287d533b082d},
                  Float64,
                  2},
                2},
              1},
            Tuple{
              UnitRange{
                Int64}},
            true}}},
      1},
1}}
```

This doesn't appear to be an issue with the previous version of `ForwardDiff`, `v0.4.2`. It may very well be an issue with `ForwardDiff` itself, but I thought I'd bring it up here first since I wasn't able to distill the example.

goretkin, 2017-09-10:
Okay... I guess I can remove a bunch of distracting code: https://gist.github.com/goretkin/da6fef8f40ec4db893b147caed2775ce

The error is still the same.

tkoolen, 2017-09-11:
Shorter:
```julia
using RigidBodyDynamics
using ForwardDiff

mechanism = rand_tree_mechanism(Revolute{Float64})

function f(x::AbstractArray{T}) where {T}
    result = DynamicsResult{T}(mechanism)
    1.
end

@show ForwardDiff.hessian(f, [10.0, 10.0, 10.0])
```

tkoolen, 2017-09-11:
Thanks. Will be fixed by #331.

closed 2017-09-11 by tkoolen

tkoolen, 2017-09-12:
By the way, last I checked, compilation times can become very long with `ForwardDiff.hessian`. You can work around this (possibly at a runtime cost) by making the chunk size smaller (e.g., 1). See https://github.com/tkoolen/RigidBodyDynamics.jl/blob/b61a157769b3e7acd103a8d32da40ffd08a75ab5/test/test_mechanism_algorithms.jl#L326-L327, although that is using the old `HessianConfig` syntax.

commit 463bb16a (2017-09-11): Fix #330 by explicitly specifying the value type of the UnsafeFastDict.
commit b61a1577 (2017-09-11): Merge pull request #331 from tkoolen/fix-330

Fix #330 by explicitly specifying the value type of the UnsafeFastDict