JuliaRobotics/RigidBodyDynamics.jl #332: mechanism made from URDF cannot be `step`d "DimensionMismatch("Lengths of input collections do not match.")"

goretkin, 2017-09-12:
```julia
using RigidBodyDynamics

urdf_text = """
<?xml version="1.0" ?>
<robot name="prism2">

  <link name="base_link">
    <inertial>
      <mass value="1.0"/>
      <origin xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1"/>
    </inertial>
  </link>

  <link name="middle_link">
    <inertial>
      <mass value="1.0"/>
      <origin xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1"/>
    </inertial>
  </link>

  <link name="tip_link">
    <inertial>
      <mass value="1.0"/>
      <origin xyz="0.0 0.0 0.0"/>
      <inertia ixx="0.1" ixy="0.0" ixz="0.0" iyy="0.1" iyz="0.0" izz="0.1"/>
    </inertial>
  </link>

  <joint name="joint1" type="prismatic">
    <axis xyz="1 0 0"/>
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.0"/>
    <parent link="base_link"/>
    <child link="middle_link"/>
  </joint>

  <joint name="joint2" type="prismatic">
    <axis xyz="1 0 0"/>
    <origin rpy="0.0 0.0 0.0" xyz="0.0 0.0 0.0"/>
    <parent link="middle_link"/>
    <child link="tip_link"/>
  </joint>
</robot>
"""

open("/tmp/urdf_unique", "w") do f
    write(f, urdf_text);
end

mechanism = parse_urdf(Float64, "/tmp/urdf_unique")
x = MechanismState(mechanism)
rand!(x)
times, qs, vs = simulate(x, 0.1; Δt = 1e-2)

"""
DimensionMismatch("Lengths of input collections do not match.")

Stacktrace:
 [1] lengths_match_fail() at /Users/goretkin/playgrounds/rbd_diff/packages/v0.6/TypeSortedCollections/src/TypeSortedCollections.jl:119
 [2] macro expansion at /Users/goretkin/playgrounds/rbd_diff/packages/v0.6/RigidBodyDynamics/src/custom_collections.jl:26 [inlined]
 [3] foreach_with_extra_args at /Users/goretkin/playgrounds/rbd_diff/packages/v0.6/RigidBodyDynamics/src/custom_collections.jl:8 [inlined]
 [4] global_coordinates!(::RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Fixed{Float64}},1},Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Prismatic{Float64}},1}},2}}, ::Array{Float64,1}, ::Array{Float64,1}) at /Users/goretkin/playgrounds/rbd_diff/packages/v0.6/RigidBodyDynamics/src/mechanism_state.jl:842
 [5] step(::RigidBodyDynamics.OdeIntegrators.MuntheKaasIntegrator{4,Float64,RigidBodyDynamics.##147#148{RigidBodyDynamics.DynamicsResult{Float64,Float64}},RigidBodyDynamics.OdeIntegrators.ExpandingStorage{Float64},16}, ::Float64, ::RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Fixed{Float64}},1},Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Prismatic{Float64}},1}},2}}, ::Float64) at /Users/goretkin/playgrounds/rbd_diff/packages/v0.6/RigidBodyDynamics/src/ode_integrators.jl:269
 """

```

By instrumenting `TypeSortedCollections.lengths_match` the failing arguments are:
```
a1 = 3
a2 = SubArray{Float64,1,Array{Float64,1},Tuple{UnitRange{Int64}},true}[Float64[], [2.30232], [-0.0928717]]
length(a1) = 1
length(a2) = 3
```
This URDF was not giving me trouble when using a master from two months ago (sorry, I lost the specific tag. I can do a bisect later today).

tkoolen, 2017-09-12:
What commit of RigidBodyDynamics is this? What is `Pkg.status("TypeSortedCollections")`?

goretkin, 2017-09-12:
oops, somehow I was not on master for `RigidBodyDynamics`, sorry!

(anyway)
```
shell> git --git-dir $(Pkg.dir("RigidBodyDynamics"))/.git rev-parse HEAD
cd58a541cf085b31913e8a22c3206fff0003a751

julia> Pkg.status("TypeSortedCollections")
 - TypeSortedCollections         0.1.1              622daf79 (dirty)
```

tkoolen, 2017-09-12:
It's still a valid issue I think; the latest published version of RigidBodyDynamics doesn't have an upper bound on TypeSortedCollections, but doesn't actually work with the latest published version of TypeSortedCollections.

tkoolen, 2017-09-12:
FTR, version 0.3.0 of RigidBodyDynamics works with version 0.1.0 of TypeSortedCollections. RigidBodyDynamics master currently works with version 0.1.1 of TypeSortedCollections.

togo59, 2017-09-19:
same problem running Pkg.test("RigidBodyDynamics")
BTW: I deleted the folder, update the cache, did  Pkg.rm("RigidBodyDynamics") followed by reinstalling and then running the above.

NB Pkg.status("TypeSortedCollections")
 - TypeSortedCollections         0.1.1

double pendulum: Error During Test
  Got an exception of type DimensionMismatch outside of a @test
  DimensionMismatch("Lengths of input collections do not match.")
  Stacktrace:
   [1] lengths_match_fail() at /home/rpowell/.julia/v0.6/TypeSortedCollections/src/TypeSortedCollections.jl:113

togo59, 2017-09-19:
Passes all tests with 0.3.0+
(Tests are awesome BTW)

tkoolen, 2017-09-19:
I'm unfortunately still waiting for a response to https://github.com/JuliaLang/METADATA.jl/pull/11158. The METADATA maintainers are typically very hesitant to add upper bounds to packages. I might just have to release a new version of RigidBodyDynamics instead.

> Tests are awesome BTW

Thank you very much.

tkoolen, 2017-09-19:
See https://github.com/JuliaLang/METADATA.jl/pull/11281.

tkoolen, 2017-09-20:
Fixed by https://github.com/JuliaLang/METADATA.jl/pull/11281.

closed 2017-09-20 by tkoolen