JuliaRobotics/RigidBodyDynamics.jl #344: RigidBodyDynamicsDiffEqIntegration package?

tkoolen, 2017-09-22:
I don't want this package to depend on DifferentialEquations.jl, but I do want it to be easy to use with DifferentialEquations. I also just want to explore DifferentialEquations a bit.

Some things that could be incorporated in such a package:
- `ODEProblem`/ `DAEProblem` constructors that take a `Mechanism`
- callback that visualizes the mechanism during the simulation using `RigidBodyTreeInspector`
- the current [`dynamics!` method that is meant for use with standard ODE solvers](https://github.com/tkoolen/RigidBodyDynamics.jl/blob/0371c566f977ad1b01a173e7a6c602d0e80d2e95/src/mechanism_algorithms.jl#L747).
- tying in to DiffEq parameter estimation? (although I think that a separate package that implements something like https://arxiv.org/pdf/1701.04395.pdf would be better, as it can exploit the structure that comes from rigid body dynamics)
- a `MuntheKaasIntegrator` wrapper that implements the `DifferentialEquations.DEIntegrator` interface?

ChrisRackauckas, 2017-09-27:
Yes, this sounds great. Let me know what you need.

>tying in to DiffEq parameter estimation? (although I think that a separate package that implements something like https://arxiv.org/pdf/1701.04395.pdf would be better, as it can exploit the structure that comes from rigid body dynamics)

Any solver on the common interface will directly work with the parameter estimation tools. There nothing else necessary there. But yes, for this case solvers which exploit the linearity and other structure of rigid body dynamics would probably do better.

tkoolen, 2017-09-27:
Here's a start: https://github.com/tkoolen/RigidBodyDynamicsDiffEqIntegration.jl. Further discussion should probably take place there.

closed 2017-09-27 by tkoolen