JuliaRobotics/RigidBodyDynamics.jl #346: Segmentation fault

phelipe, 2017-09-23:
The simulation function returns
```
signal (11): Segmentation fault'
``` 

Tests performed on two different computers.

tkoolen, 2017-09-23:
Could you provide code that exhibits the failure?

phelipe, 2017-09-23:
I used the double pendulum example and Acrobot example.

tkoolen, 2017-09-23:
Tests pass both locally (on both OSX and Linux) and on Travis, and they include various calls to `simulate`, as well as running all of the notebooks.

I'll need more to work with. I'm not clear on what code 'the double pendulum example and Acrobot example' exactly refers to. Could you post:
* the exact code that triggers the segfault
* the output of `versioninfo()`
* the output of `Pkg.status()`

phelipe, 2017-09-24:
The tests pass locally here too, however when I run any code calling the **simulate** function the error happens. An example code I used follows below, but I also tested using
https://github.com/tkoolen/RoboticsJuliaCon2017.jl/tree/master/notebook
```
using RigidBodyDynamics

urdf = "urdf/doublependulum.urdf"
mechanism = parse_urdf(Float64, urdf)
state = MechanismState{Float64}( mechanism)
times, joint_angles, joint_velocities = simulate(state, 5.)

```

```
julia> Pkg.status("RigidBodyDynamics")
 - RigidBodyDynamics             0.4.0+             master
```

```
julia> versioninfo()
Julia Version 0.6.0                                              
Commit 903644385b* (2017-06-19 13:05 UTC)                        
Platform Info:                                                   
  OS: Linux (x86_64-pc-linux-gnu)                                
  CPU: Intel(R) Core(TM) i5-4210U CPU @ 1.70GHz                  
  WORD_SIZE: 64                                                  
  BLAS: libblas                                                  
  LAPACK: liblapack                                              
  LIBM: libm                                                     
  LLVM: libLLVM-3.9.1 (ORCJIT, haswell)
```

```
julia> Pkg.status()
11 required packages:
- ControlSystems                0.3.0
- DrakeVisualizer               0.2.1+             master
- ForwardDiff                   0.5.0
- Gurobi                        0.3.3
- JuMP                          0.18.0
- LCMCore                       0.0.4              85a57d79 (dirty)
- ODE                           0.7.0
- PyPlot                        2.3.2
- RigidBodyDynamics             0.4.0+             master
- RigidBodyTreeInspector        0.3.2              master
- SymPy                         0.5.3
64 additional packages:
- AxisAlgorithms                0.2.0
- BaseTestNext                  0.2.2
- BinDeps                       0.6.0
- CMakeWrapper                  0.0.2
- Calculus                      0.2.2
- ColorTypes                    0.6.3
- Colors                        0.8.1
- CommonSubexpressions          0.0.1
- Compat                        0.32.0
- Conda                         0.7.0
- Contour                       0.4.0
- CoordinateTransformations     0.4.1
- DataStructures                0.7.2
- DiffBase                      0.2.0
- DiffEqBase                    1.23.1
- DocStringExtensions           0.4.1
- DualNumbers                   0.3.0
- FileIO                        0.5.2
- FixedPointNumbers             0.4.1
- GeometryTypes                 0.4.2
- Hiccup                        0.1.1
- IJulia                        1.6.1
- Interact                      0.6.2
- Interpolations                0.7.0
- IterTools                     0.1.0
- JSON                          0.14.0
- Juno                          0.3.1
- LaTeXStrings                  0.2.1
- LightXML                      0.5.0
- LinearMaps                    1.0.1
- MacroTools                    0.3.7
- MathProgBase                  0.6.4
- MbedTLS                       0.5.0
- Measures                      0.1.0
- Media                         0.3.0
- MeshIO                        0.1.0
- Meshing                       0.2.0
- NaNMath                       0.2.6
- Parameters                    0.7.3
- PlotThemes                    0.1.4
- PlotUtils                     0.4.4
- Plots                         0.12.4
- Polynomials                   0.1.6
- PyCall                        1.15.0
- Ratios                        0.2.0
- Reactive                      0.6.0
- RealInterface                 0.0.3
- RecipesBase                   0.2.2
- RecursiveArrayTools           0.12.3
- Reexport                      0.0.3
- Requires                      0.4.3
- ReverseDiffSparse             0.8.0
- Rotations                     0.6.0
- SHA                           0.5.1
- ShowItLikeYouBuildIt          0.1.1
- Showoff                       0.1.1
- SimpleTraits                  0.5.1
- SpecialFunctions              0.3.3
- StaticArrays                  0.6.2
- StatsBase                     0.19.0
- TypeSortedCollections         0.1.1
- URIParser                     0.2.0
- WoodburyMatrices              0.2.2
- ZMQ                           0.4.3

```

tkoolen, 2017-09-24:
Hmm, I still can't reproduce this, neither on OSX nor on Linux. To be clear, I'm on RigidBodyDynamics 0371c566f977ad1b01a173e7a6c602d0e80d2e95, and I did change `"urdf/doublependulum.urdf"` in your example code to `"urdf/Acrobot.urdf"` because I don't have a `"doublependulum.urdf"`. But you mentioned that the same thing happens with `"Acrobot.urdf"`.

I noticed in your `versioninfo` that you're using
```
  BLAS: libblas                                                  
  LAPACK: liblapack                                              
  LIBM: libm   
```
whereas I have
```
  BLAS: libopenblas (USE64BITINT DYNAMIC_ARCH NO_AFFINITY Haswell)
  LAPACK: libopenblas64_
  LIBM: libopenlibm
```
Also, your commit is `903644385b*`, whereas mine is `9036443` (no star). I'm assuming the star means local modifications that weren't committed, and that you compiled Julia from source. Could you try with the generic Linux binary downloaded from the Julia website?

By the way, the stack trace appears to be similar to https://github.com/JuliaLang/julia/issues/21244.

For completeness, here is my test setup on Linux:
```julia
julia> Pkg.status()
1 required packages:
 - RigidBodyDynamics             0.4.0+             master
7 additional packages:
 - Compat                        0.32.0
 - DocStringExtensions           0.4.1
 - LightXML                      0.5.0
 - Reexport                      0.0.3
 - Rotations                     0.6.0
 - StaticArrays                  0.6.2
 - TypeSortedCollections         0.1.1
```
(same dependency versions as you were using)

```julia
julia> versioninfo()
Julia Version 0.6.0
Commit 9036443 (2017-06-19 13:05 UTC)
Platform Info:
  OS: Linux (x86_64-pc-linux-gnu)
  CPU: Intel(R) Core(TM) i7-6950X CPU @ 3.00GHz
  WORD_SIZE: 64
  BLAS: libopenblas (USE64BITINT DYNAMIC_ARCH NO_AFFINITY Haswell)
  LAPACK: libopenblas64_
  LIBM: libopenlibm
  LLVM: libLLVM-3.9.1 (ORCJIT, broadwell)
```

tkoolen, 2017-09-27:
Thank you very much for the update. Just so that I understand fully, was this because you compiled Julia against a vanilla version of LLVM that didn't include the patches from https://github.com/JuliaLang/julia/pull/19810?

phelipe, 2017-09-28:
use Manjaro (based on Arch linux), julia is in the package manager of it. However since the Arch is rolling-release system, the error may have occurred after a system update. Some people reported that same problem on Slack, including problems similar to the ones I had when I installed DrakeVisualizer.