JuliaRobotics/RigidBodyDynamics.jl #347: JuMP autodiff of `dynamics!` not working

goretkin, 2017-09-23:
The same code which works in a global scope does not work inside a function scope. Does that suggest another type-inference bug?

```julia
using RigidBodyDynamics
using JuMP
using Ipopt

# Choose either one of these and get a different error, 
# mechanism = rand_tree_mechanism(Revolute{Float64})
mechanism = rand_tree_mechanism(Revolute{Float64}, Revolute{Float64})

gstate = MechanismState{Float64}(mechanism)
zero!(gstate)

function f(state, τ_solve::AbstractArray{T}) where T
    dynamics_result = DynamicsResult{T}(mechanism)
    state_copy = MechanismState{T}(state.mechanism)
    
    RigidBodyDynamics.dynamics!(dynamics_result, state_copy, τ_solve)
    return dynamics_result.v̇[1]
end

function optimize(state)
    m = Model(solver = IpoptSolver(print_level=0))
    JuMP.register(m, :jf, num_velocities(mechanism),
        (τ...) -> begin f(state, [τ...]) end,
        autodiff=true)
    
    @variable(m, -1.0 <= τ[1:num_velocities(mechanism)] <= 1.0)
    # @NLobjective(m, Min, jf(τ...))
    JuMP.setNLobjective(m, :Min, Expr(:call, :jf, [τ[i] for i=1:length(τ)]...))
    solve(m)
    getvalue(τ)
end

# doesn't work
optimize(gstate)

# this works fine
begin
    state = gstate # bind formal parameter
    
    # function body of `optimize`
    m = Model(solver = IpoptSolver(print_level=0))
    JuMP.register(m, :jf, num_velocities(mechanism),
        (τ...) -> begin f(state, [τ...]) end,
        autodiff=true)

    @variable(m, -1.0 <= τ[1:num_velocities(mechanism)] <= 1.0)
    # @NLobjective(m, Min, jf(τ...))
    JuMP.setNLobjective(m, :Min, Expr(:call, :jf, [τ[i] for i=1:length(τ)]...))
    solve(m)
    getvalue(τ)
end
```

The case where the mechanism has only joint appears to get a bit farther (it calls ForwardDiff's derivative, instead of ForwardDiff's gradient.)

Gradient error:
```
The provided configuration (of type ForwardDiff.GradientConfig{ForwardDiff.Tag{JuMP.##166#168{##11#13{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{JuMP.##166#168{##11#13{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,2},1}}) was constructed for a function other than the current target function. ForwardDiff cannot safely perform differentiation in this context; see the following issue for details: https://github.com/JuliaDiff/ForwardDiff.jl/issues/83. You can resolve this problem by constructing and using a configuration with the appropriate target function, e.g. `ForwardDiff.GradientConfig(JuMP.#166, x)`
```

Derivative error:
```
MethodError: no method matching setindex!(::RigidBodyDynamics.CustomCollections.UnsafeFastDict{RigidBodyDynamics.Graphs.vertex_index,RigidBodyDynamics.RigidBody{Float64},RigidBodyDynamics.Spatial.Transform3D{ForwardDiff.Dual{ForwardDiff.Tag{##2#4{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}},0x71fee09dfcb49691},ForwardDiff.Dual{ForwardDiff.Tag{JuMP.##171#173{##2#4{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,1},1}}}, ::RigidBodyDynamics.Spatial.Transform3D{Any}, ::RigidBodyDynamics.RigidBody{Float64})
```

(the `setindex!` error is that the value has type `::RigidBodyDynamics.Spatial.Transform3D{Any}`, but the value type inside the `UnsafeFastDict` is `::RigidBodyDynamics.Spatial.Transform3D{lots_of_forward_diff_type_parameters}`)

Full stack trace and package versions here: 
https://gist.github.com/4530d780a6650a572b355b910adc3bb8

tkoolen, 2017-09-23:
You're on ForwardDiff master and I can't see the commit SHA, so I can't reproduce this. With JuMP 0.18.0, ForwardDiff 0.5.0, and RigidBodyDynamics at 0371c566f977ad1b01a173e7a6c602d0e80d2e95, I get a different error:
```
UndefVarError: ConfigHismatchError not defined

Stacktrace:
 [1] (::JuMP.##167#169{JuMP.##166#168{##7#9{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},ForwardDiff.GradientConfig{ForwardDiff.Tag{JuMP.##166#168{##7#9{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,2,Array{ForwardDiff.Dual{ForwardDiff.Tag{JuMP.##166#168{##7#9{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,2},1}}})(::SubArray{Float64,1,Array{Float64,1},Tuple{UnitRange{Int64}},true}, ::SubArray{Float64,1,Array{Float64,1},Tuple{UnitRange{Int64}},true}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:1365
 [2] eval_grad_f(::JuMP.UserFunctionEvaluator, ::SubArray{Float64,1,Array{Float64,1},Tuple{UnitRange{Int64}},true}, ::SubArray{Float64,1,Array{Float64,1},Tuple{UnitRange{Int64}},true}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:1358
 [3] #forward_eval#7(::ReverseDiffSparse.UserOperatorRegistry, ::Function, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{ReverseDiffSparse.NodeData,1}, ::SparseMatrixCSC{Bool,Int64}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/ReverseDiffSparse/src/forward.jl:129
 [4] (::ReverseDiffSparse.#kw##forward_eval)(::Array{Any,1}, ::ReverseDiffSparse.#forward_eval, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{ReverseDiffSparse.NodeData,1}, ::SparseMatrixCSC{Bool,Int64}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}) at ./<missing>:0
 [5] forward_eval_all(::JuMP.NLPEvaluator, ::Array{Float64,1}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:447
 [6] eval_grad_f(::JuMP.NLPEvaluator, ::Array{Float64,1}, ::Array{Float64,1}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:500
 [7] initialize(::JuMP.NLPEvaluator, ::Array{Symbol,1}) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:405
 [8] loadproblem!(::Ipopt.IpoptMathProgModel, ::Int64, ::Int64, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Array{Float64,1}, ::Symbol, ::JuMP.NLPEvaluator) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/Ipopt/src/IpoptSolverInterface.jl:44
 [9] _buildInternalModel_nlp(::JuMP.Model, ::JuMP.ProblemTraits) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/nlp.jl:1248
 [10] #build#119(::Bool, ::Bool, ::JuMP.ProblemTraits, ::Function, ::JuMP.Model) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/solvers.jl:303
 [11] (::JuMP.#kw##build)(::Array{Any,1}, ::JuMP.#build, ::JuMP.Model) at ./<missing>:0
 [12] #solve#116(::Bool, ::Bool, ::Bool, ::Array{Any,1}, ::Function, ::JuMP.Model) at /Users/twan/code/julia/RigidBodyDynamics/v0.6/JuMP/src/solvers.jl:168
 [13] optimize(::RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}) at ./In[5]:29
 [14] include_string(::String, ::String) at ./loading.jl:515
```
which appears to be a typo in ForwardDiff that was fixed on a more recent version.

What sha of ForwardDiff are you on?

goretkin, 2017-09-23:
Yeah, I am on a master of `ForwardDiff` so that it produces a slightly more meaningful error
Sorry the commit hash was missing, it's https://github.com/JuliaDiff/ForwardDiff.jl/commit/887d282b224033778cfdbfb91b35d4fddd785fe9

There's one commit after that, but it depends on https://github.com/JuliaDiff/DiffResults.jl which doesn't appear to be tagged yet.

tkoolen, 2017-09-24:
Thanks. Still looking into this.

Just so you know, if you really just want the gradient of the joint accelerations w.r.t. the torques, you don't really need ForwardDiff for that (and the computation will be much more efficient without it in this case). It's particularly easy for mechanisms without kinematic loops, for which the joint accelerations are simply:

[image]

so the gradient w.r.t. tau is just [image] (see `dynamics!` documentation for notation).

goretkin, 2017-09-24:
I'm interested to know what the fix is. I went hunting, but unfortunately I'm not familiar enough with ...many things... to be effective. If you have a lead you think I could go of off, I wouldn't mind hearing it.

Thanks for the alternative. In this case, I'm actually trying to take derivatives of the kinetic energy at the end of an integration step (to implement Coulomb friction via the maximum dissipation principle). I had mentioned that I had it working (albeit inefficiently) at some point and now I'm trying to get it working again with the latest versions of the packages.

tkoolen, 2017-09-24:
Well, by changing the `*` method for `Transform3D` args to
```julia
@inline function Base.:*(t1::Transform3D, t2::Transform3D)
    @framecheck(t1.from, t2.to)
    mat = t1.mat * t2.mat
    if eltype(mat) == Any
        @show eltype(t1)
        @show eltype(t2)
        @show Base.promote_op(Base.LinAlg.matprod, eltype(t1), eltype(t2))
    end
    Transform3D(t2.from, t1.to, mat)
end
```

I did figure out that for
```julia
eltype(t1) = Float64
eltype(t2) = ForwardDiff.Dual{ForwardDiff.Tag{##3#5{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}},0x2e6b9ae0e43b2cc6},ForwardDiff.Dual{ForwardDiff.Tag{JuMP.##171#173{##3#5{RigidBodyDynamics.MechanismState{Float64,Float64,Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{RigidBodyDynamics.Joint{Float64,RigidBodyDynamics.Revolute{Float64}},1}},1}}}},0xb046287d533b082d},Float64,1},1}
```
`eltype(mat)` becomes `Any` (as does `Base.promote_op(Base.LinAlg.matprod, eltype(t1), eltype(t2))
`, which is [used in StaticArrays to determine the type of the returned matrix](https://github.com/JuliaArrays/StaticArrays.jl/blob/7cc7f9dee3a3524f4a0e4f4b804d80c713669c6f/src/matrix_multiply.jl#L171)). This may just be inference giving up. Note that this is a nested `ForwardDiff.Dual` (this is a second derivative evaluation). Maybe it is a problem that the `MechanismState` type (which is 'complicated') ends up in the `Dual`'s tag parameter. I believe this happens as a result of a closure over the `MechanismState` in the function you `Jump.register` (although you do the same thing in the example that does work)...

If you want to help out, you could:
* try to find a smaller example, perhaps involving only ForwardDiff, StaticArrays, and a sufficiently complex type appearing in the `Dual`'s tag, that shows the same behavior
* investigate further why the `promote_op` call results in `Any`
* investigate why the second example *does* work by looking at the types with which the `Transform3D` `*` method gets called in that case.

tkoolen, 2017-09-25:
As for 3. above: for that second example, the `Dual` types look more like 
```julia
ForwardDiff.Dual{ForwardDiff.Tag{##7#9,0x4eb83fe739c3f060},ForwardDiff.Dual{ForwardDiff.Tag{JuMP.##171#173{##7#9},0xb046287d533b082d},Float64,1},1}
```
i.e. the `MechanismState` type doesn't show up in the `Dual` type. That seems to align with my theory that the `MechanismState` type parameter is making the `Dual` type too complicated, causing inference to give up.

tkoolen, 2017-09-25:
OK, I've verified that the gradient error above (for the two-joint mechanism) is fixed by https://github.com/JuliaDiff/ForwardDiff.jl/pull/247 (it's a case of perturbation confusion). I submitted a PR against that branch that adds a test that is representative of the case we encountered here. The error with the derivative method remains.

goretkin, 2017-09-25:
Wow, thanks for finding that test. Did you just make more and more complicated `z83a` until the issue was triggered?

It looks like, as mentioned here, https://github.com/JuliaDiff/ForwardDiff.jl/pull/247#issuecomment-331941842 your test nicely demonstrates that the issue is sensitive to scope.

tkoolen, 2017-09-25:
> Did you just make more and more complicated z83a until the issue was triggered?

Yep.

tkoolen, 2018-01-18:
Even though the ForwardDiff issue is still open, the problem in the issue description appears to be solved with ForwardDiff 0.7.0, so I'm going to close this.

closed 2018-01-18 by tkoolen