JuliaRobotics/RigidBodyDynamics.jl #363: Conveniently avoiding allocating new MechanismState objects

rdeits, 2017-11-17:
I've found in my code that I end up allocating a lot of extra MechanismState objects at various points in the code because it may be inconvenient to pass around a single state, or because I don't necessarily know the scalar type that a given function will be using. For example, if I want to compute the gradient of some function `f(x::MechanismState)`, then I might do it like this:

```julia
# assume we have x::MechanismState{Float64, ...}
g = ForwardDiff.gradient(state_vector(x)) do x_vector
  x_dual = MechanismState(x.mechanism, x_vector) # x_dual will have scalar type Dual{N, Float64}, but ForwardDiff is going to pick N for me
  f(x_dual)
end
```

This wastes a lot of time by constructing a new MechanismState every time the inner function is called (which might be multiple times for chunk-mode AD. Unfortunately, it's tricky to pre-allocate a state because to do something like:

```julia
x_dual = MechanismState{Dual{N, Float64}(mechanism)
g = ForwardDiff.gradient(state_vector(x)) do x_vector
  set_state!(x_dual, x_vector)
  f(x_dual)
end
```

I have to know the value of `N` ahead of time. Even if I did, this is still wasteful unless I keep `x_dual` around and re-use it every time I need to do some AD. 

One way to avoid this would be to create some kind of cache of MechanismStates. Ideally, I would have some function `scratch_state(mechanism, scalartype)` that would give me a MechanismState of the given type, but would only construct a single instance for each scalar type and mechanism. 

### Proposal 1: per-mechanism StateCache:

Here's one possibility:

```julia
struct StateCache{M, JointCollection}
    mechanism::Mechanism{M}
    states::Dict{Tuple{Int, DataType}, MechanismState}
end

function StateCache(mechanism::Mechanism{M}) where M
    JointCollection = typeof(TypeSortedCollection(typedjoint.(joints(mechanism))))
    StateCache{M, JointCollection}(mechanism, Dict())
end

function Base.getindex(c::StateCache{M, JC}, ::Type{X}) where {M, JC, X}
    C = promote_type(X, M)
    get!(c.states, (Threads.threadid(), X)) do
        MechanismState{X}(c.mechanism)
    end::MechanismState{X, M, C, JC}
end
```

You would use this as:

```julia
c = StateCache(mechanism)
c[Float64] # this is type-stable because the type of `c` includes all the JointCollection type info
```

This generally solves the problem, but it isn't perfect:

* Lookup in the cache takes ~70ns and 2 allocations
* Users always have to pass around the cache object, which is only slightly easier than just passing around `x_dual`

It would be interesting to have a global state cache that maps `(mechanism, scalar) -> MechanismState`, but there's a problem with that: The type of a MechanismState cannot be determined from just the type of the Mechanism and the scalar because it also includes the particular JointCollection. That means that lookup into the state cache by mechanism and scalar cannot be type-stable. 

@tkoolen any thoughts? Is there some other way of approaching the problem?

A different, more invasive, proposal would be to create some kind of TypedMechanism, which would include the JointCollection in its type. A MechanismState could be derived from a TypedMechanism in a type-inferable way, and this might also resolve the current issue that I can create a MechanismState, then modify the mechanism's joints, and render that state invalid. 

Having a typed mechanism, and being able to construct a state from that typed mechanism, would mean that we could simply do:

```julia
@generated function scratch_state(::M, Scalar) where {M <: TypedMechanism, Scalar}
  state = MechanismState{M, Scalar}() # construct a state in the generator body (this will be done once)
  state # return that singleton state, so future calls to this function get the *same* state object
end
```

As you mentioned, @tkoolen, there's no guarantee that the state would actually be a singleton, since the generator can be called multiple times. That's fine, though: the singleton-ness is just an optimization.

rdeits, 2017-11-17:
The idea with Proposal 2 is that you'd treat Mechanism as a container for making modifications to the system, and then you'd solidify that into an (immutable) TypedMechanism when you wanted to actually use it to generate states, etc.

rdeits, 2017-11-17:
Another option:

### Proposal 3: Construct new states from an existing state

This is essentially the same as Proposal 2, except that we just treat the MechanismState as the TypedMechanism. So we would write a function like this:

```julia
@generated function similar_state(::M, Scalar) where {M <: MechanismState, Scalar}
  state = MechanismState{M, Scalar}() # hypothetical constructor
  state
end
```

This would require writing a MechanismState constructor that takes an existing MechanismState and a new scalar type, but I think that would be super easy, right? 

Proposal 3 requires that the user pass around a MechanismState everywhere, but that's totally reasonable, since it's required to do anything at all.

tkoolen, 2017-11-20:
With proposal 2 and 3, I'm worried about world age issues and in general the fact that it's not the normal way generated functions are supposed to be used. I believe there are still more changes coming to generated functions, so I'm hesitant to go that route even if it does work right now. I also just think that a `Dict`-like object is a better solution.

Here's a slightly modified version of proposal 1 which doesn't allocate (avoids creating a tuple containing a `DataType` and a closure over a `DataType` and `Mechanism`) and takes 50 ns for lookup (compared to 63 ns on my machine for the original proposal):
```julia
struct StateCache{M, JointCollection}
    mechanism::Mechanism{M}
    states::Dict{UInt64, MechanismState}
end

function StateCache(mechanism::Mechanism{M}) where M
    JointCollection = typeof(TypeSortedCollection(typedjoint.(joints(mechanism))))
    StateCache{M, JointCollection}(mechanism, Dict())
end

@inline function Base.getindex(c::StateCache{M, JC}, ::Type{X}) where {M, JC, X}
    C = promote_type(X, M)
    key = hash(Threads.threadid(), hash(X))
    if haskey(c.states, key)
        c.states[key]
    else
        c.states[key] = MechanismState{X}(c.mechanism)
    end::MechanismState{X, M, C, JC}
end
```
Would that be good enough for you?

By the way,

> and this might also resolve the current issue that I can create a MechanismState, then modify the mechanism's joints, and render that state invalid

is not true, because the state becomes invalid also if a new joint of a type that already existed in the `Mechanism` is added (but this is handled by a modcount check).

tkoolen, 2017-11-20:
Or with an O(n) lookup, but faster (22 ns) when there's only a few different types (the most common use case):
```julia
struct StateCache{M, JointCollection}
    mechanism::Mechanism{M}
    keys::Vector{UInt64}
    states::Vector{MechanismState}
end

function StateCache(mechanism::Mechanism{M}) where M
    JointCollection = typeof(TypeSortedCollection(typedjoint.(joints(mechanism))))
    StateCache{M, JointCollection}(mechanism, UInt64[], MechanismState[])
end

@inline function Base.getindex(c::StateCache{M, JC}, ::Type{X}) where {M, JC, X} 
    C = promote_type(X, M)
    ReturnType = MechanismState{X, M, C, JC}
    key = hash(Threads.threadid(), hash(X))
    for i = 1 : length(c.keys)
        if c.keys[i] == key
            return c.states[i]::ReturnType
        end
    end
    push!(c.keys, key)
    state = MechanismState{X}(c.mechanism)
    push!(c.states, state)
    state::ReturnType
end
```

rdeits, 2017-11-21:
Yeah, I agree that the generated trick is a little too magical. Probably better to use `object_id` directly rather than hash(), since `a == b` implies `hash(a) == hash(b)`, but the inverse is not true. For `Type`s and `Mechanism`s I think object_id should always be the right thing to do.

tkoolen, 2017-11-21:
Alright, I guess hash collisions are not impossible (just supposed to be extremely unlikely). Also shaves off another 4 ns.

rdeits, 2017-11-25:
oh, whoops, I thought #365 was already in. I'll just let this get closed normally by the PR

PR #365 (2017-11-26): Add StateCache
CC: @rdeits.

Fixes #363.
closed 2017-11-26 by tkoolen