JuliaRobotics/RigidBodyDynamics.jl #425: New release?

rdeits, 2018-04-24:
It would be great to have the planar joint URDF parsing in a released version so I can depend on it in atlas-distro. Is this a good time for 0.5.1?

tkoolen, 2018-04-25:
OK. It would be 0.6 according to semver though.

tkoolen, 2018-04-25:
Are all your packages already working with RigidBodyDynamics master?

rdeits, 2018-04-25:
Successfully tested: 

- [-] MechanismGeometries
- [-] MeshCatMechanisms
- [-] RigidBodyTreeInspector
- [-] LCPSim

tkoolen, 2018-04-25:
Great, thank you!

To do:

* [ ] Run tests from previous released version against master. Ensure that there are reasonable deprecation messages, etc.
* [ ] Run tests for registered dependencies:
  * [x] MechanismGeometries
  * [x] MeshCatMechanisms
  * [x] RigidBodyTreeInspector
  * [ ] MotionCaptureJointCalibration
  * [ ] RigidBodySim
  * [ ] ValkyrieRobot
  * [ ] AtlasRobot
* Run tests for unregistered dependencies:
  * [x] LCPSim
  * [ ] HumanoidLCMSim
* [ ] Ensure that docs are up to date and that they build properly
* [ ] Update benchmark results, if relevant
* [ ] Use Attobot to set up release
* [ ] Update `NEWS.md`

closed 2018-04-26 by tkoolen