JuliaRobotics/RigidBodyDynamics.jl #429: URDF rpy parsing might be incorrect

rdeits, 2018-05-11:
I just checked against ROS's `tf.transformations` module, and it *appears* that to match the behavior of ROS's `rpy` we have to do `RotZYX(y, p, r)`. 

```python
>>> tf.transformations.quaternion_matrix(tf.transformations.quaternion_from_euler(1, 2, 3))
array([[ 0.41198225, -0.83373765, -0.36763046,  0.        ],
       [-0.05872664, -0.42691762,  0.90238159,  0.        ],
       [-0.90929743, -0.35017549, -0.2248451 ,  0.        ],
       [ 0.        ,  0.        ,  0.        ,  1.        ]])
```

```julia
julia> RotZYX(3, 2, 1)
3×3 RotZYX{Int64}(3, 2, 1):
  0.411982   -0.833738  -0.36763 
 -0.0587266  -0.426918   0.902382
 -0.909297   -0.350175  -0.224845
```

rdeits, 2018-05-11:
From the Rotations.jl manual: 

> if you prefer the "intrinsic" form it is easy to use the corresponding RotZYX(z, y, x)

tkoolen, 2018-05-11:
Yep, as discussed in person. The *one* place RPY is used it's wrong. Thanks for the fix.

PR #430 (2018-05-11): fix parsing of ROS-formatted rpy in URDFs and add some tests
Fixes #429
closed 2018-05-11 by tkoolen