JuliaRobotics/RigidBodyDynamics.jl #434: Array conversions consistency

rdeits, 2018-06-13:
To get a flat state vector from a `MechanismState`, we do `Vector(state)`, but to get a plain matrix from a GeometricJacobian, we do `Array(J)` not `Matrix(J)`. It would be nice to make those consistent. I'd suggest allowing `Matrix(J)` and possibly deprecating `Array(J)` in the future. This also comes up with `SpatialInertia`, `MomentumMatrix`, etc.

tkoolen, 2018-06-18:
I guess Base allows `Array(A)`, `Matrix(A)`, and even `Array{Float64}(A)` and `Matrix{Float64}(A)`, so I'm not sure about deprecating `Array(J)`. But I agree this should be made more consistent.

tkoolen, 2018-10-08:
Using SparseArrays as an example, this all works:

```julia
using SparseArrays
x = spzeros(Float32, 4)
Vector(x)
Vector{Float64}(x
Array(x)
Array{Float64}(x)
```

commit 19567988 (2018-10-20): Merge pull request #498 from JuliaRobotics/tk/fix-array-conversion

Fix #434, array conversion consistency.
commit e4084457 (2018-10-20): Fix #434, array conversion consistency.

Use SparseArrays as a model (see
https://github.com/JuliaRobotics/RigidBodyDynamics.jl/issues/434#issuecomment-427955966).

Also reduce code duplication a bit.
closed 2018-10-20 by tkoolen