JuliaRobotics/RigidBodyDynamics.jl #441: DiscardVector can't be `display()`ed

rdeits, 2018-06-20:
e.g. 

```julia
julia> RigidBodyDynamics.DiscardVector(5)
5-element RigidBodyDynamics.CustomCollections.DiscardVector:
Error showing value of type RigidBodyDynamics.CustomCollections.DiscardVector:
ERROR: indexing not defined for RigidBodyDynamics.CustomCollections.DiscardVector
Stacktrace:
 [1] getindex at ./abstractarray.jl:882 [inlined]
 [2] _getindex at ./abstractarray.jl:924 [inlined]
 [3] getindex(::RigidBodyDynamics.CustomCollections.DiscardVector, ::Int64, ::Int64) at ./abstractarray.jl:883
 [4] isassigned(::RigidBodyDynamics.CustomCollections.DiscardVector, ::Int64, ::Int64, ::Vararg{Int64,N} where N) at ./abstractarray.jl:222
 [5] alignment(::IOContext{Base.Terminals.TTYTerminal}, ::RigidBodyDynamics.CustomCollections.DiscardVector, ::Base.OneTo{Int64}, ::Base.OneTo{Int64}, ::Int64, ::Int64, ::Int64) at ./show.jl:1353
 [6] print_matrix(::IOContext{Base.Terminals.TTYTerminal}, ::RigidBodyDynamics.CustomCollections.DiscardVector, ::String, ::String, ::String, ::String, ::String, ::String, ::Int64, ::Int64) at ./show.jl:1482
 [7] print_matrix(::IOContext{Base.Terminals.TTYTerminal}, ::RigidBodyDynamics.CustomCollections.DiscardVector, ::String, ::String, ::String) at ./show.jl:1455
 [8] #showarray#265(::Bool, ::Function, ::IOContext{Base.Terminals.TTYTerminal}, ::RigidBodyDynamics.CustomCollections.DiscardVector, ::Bool) at ./show.jl:1704
 [9] display(::Base.REPL.REPLDisplay{Base.REPL.LineEditREPL}, ::MIME{Symbol("text/plain")}, ::RigidBodyDynamics.CustomCollections.DiscardVector) at ./REPL.jl:122
 [10] display(::Base.REPL.REPLDisplay{Base.REPL.LineEditREPL}, ::RigidBodyDynamics.CustomCollections.DiscardVector) at ./REPL.jl:125
 [11] display(::RigidBodyDynamics.CustomCollections.DiscardVector) at ./multimedia.jl:218
 [12] eval(::Module, ::Any) at ./boot.jl:235
 [13] print_response(::Base.Terminals.TTYTerminal, ::Any, ::Void, ::Bool, ::Bool, ::Void) at ./REPL.jl:144
 [14] print_response(::Base.REPL.LineEditREPL, ::Any, ::Void, ::Bool, ::Bool) at ./REPL.jl:129
 [15] (::Base.REPL.#do_respond#16{Bool,Base.REPL.##26#36{Base.REPL.LineEditREPL,Base.REPL.REPLHistoryProvider},Base.REPL.LineEditREPL,Base.LineEdit.Prompt})(::Base.LineEdit.MIState, ::Base.AbstractIOBuffer{Array{UInt8,1}}, ::Bool) at ./REPL.jl:646
```

We should just overload `show()` so it doesn't try to print the elements.

rdeits, 2018-06-20:
Actually, array printing is a bit more complicated. One option would be to define `getindex` to always throw an `UndefRefError`: 

```julia
Base.getindex(v::DiscardVector, i::Int) = throw(UndefRefError())
```

which automatically gives us: 

```julia
julia> RigidBodyDynamics.DiscardVector(5)
5-element RigidBodyDynamics.CustomCollections.DiscardVector:
 #undef
 #undef
 #undef
 #undef
 #undef
```

tkoolen, 2018-06-24:
Long term plan is to get rid of this again, and just use `foreach`; I still need to test if this is possible without allocating on 0.7.

tkoolen, 2018-07-09:
I think you probably opened this issue because `dynamics_bias!` accidentally returned a `DiscardVector`, is that right? That's fixed now. `DiscardVector` is really an implementation detail, so I don't want to build up a lot of functionality around it, especially since I'd rather see it go away asap. I'm going to close this as a wontfix.

closed 2018-07-09 by tkoolen

rdeits, 2018-07-09:
Actually it was just because I was playing around with broadcasting over a TypeSortedCollection. I'm fine with not fixing this.