JuliaRobotics/RigidBodyDynamics.jl #446: Last release to support 0.6

tkoolen, 2018-07-03:
To do:

* [ ] check REQUIRE files
* [ ] Run tests from previous released version against master. Ensure that there are reasonable deprecation messages, etc.
* Run tests for registered dependencies:
  * [ ] MechanismGeometries
  * [ ] MeshCatMechanisms
  * [ ] RigidBodyTreeInspector
  * [ ] MotionCaptureJointCalibration
  * [ ] RigidBodySim
  * [ ] ValkyrieRobot
  * [ ] AtlasRobot
* Run tests for unregistered dependencies:
  * [ ] LCPSim
  * [ ] HumanoidLCMSim
* [ ] Ensure that docs are up to date and that they build properly
* [ ] Update benchmark results, if relevant
* [ ] Use Attobot to set up release
* [ ] Update `NEWS.md`

tkoolen, 2018-07-03:
Release notes:

* Compatibility with Julia 0.7-beta (save for some notebooks due to additional test dependencies), with some deprecation warnings
* Store motion subspaces and wrench subspaces column by column in `MechanismState`. This allows dropping two type parameters from `MechanismState` (`MotionSubspaceCollection` and `WrenchSubspaceCollection`). See https://github.com/JuliaRobotics/RigidBodyDynamics.jl/pull/435 for more info.
* Add `point_jacobian!`/`point_jacobian` (thanks, @rdeits!)
* Implement Jacobians for q̇ <-> v (thanks, @rdeits!)

rdeits, 2018-07-03:
No problem. That package is still in research-code-land and I'm only bothering to keep it working with RBD master.

tkoolen, 2018-07-03:
But this was with latest master? Anyway, https://github.com/JuliaLang/METADATA.jl/pull/15493.

rdeits, 2018-07-03:
Oh, I see what's going on. It's calling the in-place per-joint method, but we got rid of that method during the updates to #440 when we switched them to return static arrays instead. Anyway, i'll make sure it works with master again.

PR #455 (2018-07-09): Update news after 0.7 release
Close #446.
closed 2018-07-09 by tkoolen