JuliaRobotics/RigidBodyDynamics.jl #480: Constraint drift in a complex linkage

rdeits, 2018-08-28:
So I was feeling a bit inspired by #478 and decided to take a brief break from thesis-ing :wink: 

https://www.youtube.com/watch?v=rvhsje29Alg&feature=youtu.be

All the relevant code is here: https://github.com/rdeits/StrandbeestRobot.jl/blob/master/Strandbeest%20simulation.ipynb , including generating the non-tree joints by parsing the `loop_joint` tags in the URDF and solving for a consistent initial configuration using Optim.jl. 

Overall it was pretty painless: the nonlinear solve with Optim seems to work fine, and once I got the cost function right I didn't have any issues with local minima. It takes a couple of seconds to find the initial configuration on my laptop--I'm sure we could do better, but this is pretty good. 

Simulation works fine, but only up to a point. The constraints still exponentially drift apart, and the mechanism basically falls apart (and starts returning NaN states) after a little more than 4 seconds. 

But still, this is pretty cool and I think it would be great to have this as a fully working demo.

tkoolen, 2018-08-28:
Awesome! I'll take look soonish.

rdeits, 2018-08-28:
Potentially relevant: most of the constraints seem to be perfectly stable over the course of the simulation. In fact, for every leg, there is exactly one joint that comes uncoupled during simulation, and it's always the loop joint between bars b/d/e (which for a solid unit) and bar c: https://files.tested.com/photos/2015/04/05/74549-leg-measurement.jpg . In the URDF, those are the joints with names `pairXX_legX_loop_b_c`. Notably, those are the only parts of the mechanism where a single link (`bar_c`) is the successor to *two* different loop joints (`loop_b_c` and `loop_a_c`).

tkoolen, 2018-09-03:
Where is the URDF?

tkoolen, 2018-09-03:
Ah, I guess just https://raw.githubusercontent.com/RobotLocomotion/drake/last_sha_with_original_matlab/drake/examples/Strandbeest/Strandbeest.urdf.

rdeits, 2018-09-03:
It's in the same repo as that notebook: https://github.com/rdeits/StrandbeestRobot.jl/tree/master/data

tkoolen, 2018-09-03:
Why is link 'k' a child of link 'j' in the tree? I would have expected it to be a child of link 'm'.

rdeits, 2018-09-03:
I don't know. Probably an artifact of my attempt to keep as much of the complexity as possible contained within the single-leg .xacro file. Links `a`, `l`, and `m` have to be shared between a pair of legs, so all the joints involving them have to be defined in https://github.com/rdeits/StrandbeestRobot.jl/blob/master/data/LegPair.xacro instead of https://github.com/rdeits/StrandbeestRobot.jl/blob/master/data/LegAssembly.xacro . It should be perfectly possible to change that arrangement, though.

tkoolen, 2018-09-03:
Seems to be a problem with the proportional term. Setting the P gain to zero makes this work pretty well.

rdeits, 2018-09-03:
Wow, that's astonishing. I'll update the notebook file

PR #482 (2018-09-04): Fixes related to loop joint functionality.
* fix failure to propagate stabilization_gains kwarg in a constraint_bias!
method
* fix linearized pd control method: position error needs to be rotated
to body frame (same as in nonlinear method)
* fix default for flipped_joint_map kwarg in remove_joint!

Fixes #480.
closed 2018-09-04 by tkoolen