JuliaRobotics/RigidBodyDynamics.jl #483: Mechanisms with contact points cannot be used with dynamics result caches and ForwardDiff

rdeits, 2018-09-04:
For example, trying to run the initial state optimization from #480 on a mechanism with any contact points results in: 

```julia
MethodError: Cannot `convert` an object of type ContactPoint{Float64,SoftContactModel{HuntCrossleyModel{Float64},ViscoelasticCoulombModel{Float64}}} to an object of type ContactPoint{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10},SoftContactModel{HuntCrossleyModel{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10}},ViscoelasticCoulombModel{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10}}}}
Closest candidates are:
  convert(::Type{T}, !Matched::T) where T at essentials.jl:154
  ContactPoint{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10},SoftContactModel{HuntCrossleyModel{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10}},ViscoelasticCoulombModel{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10}}}}(::Any, !Matched::Any) where {T, M<:SoftContactModel} at /home/rdeits/.julia/packages/RigidBodyDynamics/Fu7Ck/src/contact.jl:96

Stacktrace:
 [1] DynamicsResult{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10},M} where M(::Mechanism{Float64}) at /home/rdeits/.julia/packages/RigidBodyDynamics/Fu7Ck/src/dynamics_result.jl:64
 [2] makevalue at /home/rdeits/.julia/packages/RigidBodyDynamics/Fu7Ck/src/caches.jl:84 [inlined]
 [3] getindex(::DynamicsResultCache{Float64}, ::Type{ForwardDiff.Dual{ForwardDiff.Tag{getfield(Main, Symbol("##3#4")){StateCache{Float64,TypeSortedCollections.TypeSortedCollection{Tuple{Array{Joint{Float64,Fixed{Float64}},1},Array{Joint{Float64,Revolute{Float64}},1}},2}},DynamicsResultCache{Float64}},Float64},Float64,10}}) at /home/rdeits/.julia/packages/RigidBodyDynamics/Fu7Ck/src/caches.jl:13
```

rdeits, 2018-09-05:
The issue is here: https://github.com/JuliaRobotics/RigidBodyDynamics.jl/blob/185a11a84c6c2e855588c5953dd2e00a79003caf/src/dynamics_result.jl#L64

I'm not sure why that type annotation is present at all, but I think it should be `::DefaultContactPoint{M}`, not `T`.

tkoolen, 2018-09-05:
Thanks for digging in! I can't remember why I did that; I'll look through the commit messages, but it can probably just go away.

PR #485 (2018-09-07): fix DynamicsCache lookup for mechanisms with contact points
Fixes #483 

I can also just remove the annotation, but on the assumption that it was put there for a reason, I figured I would just fix it.
closed 2018-09-07 by tkoolen