JuliaRobotics/RigidBodyDynamics.jl #496: URDF generated objects not in scope?

togo59, 2018-10-08:
No problem generating Mechanisms manually and using set_configutation!() etc. But if I use the following code:
```
using RigidBodyDynamics
using LinearAlgebra
using StaticArrays
#
filename = "/home/rpowell/Documents/Julia/APCS/TARM/URDF/TARM.urdf"
boom = parse_urdf(Float64, filename) #Calc
```
which returns the expect tree (below) but it seems the kinematic tree objects are not in scope. E.g.:
```
Spanning tree:
Vertex: world (root)
  Vertex: mast, Edge: mast_to_world
    Vertex: a1_actuator_pt1, Edge: mast_a1_actuator_pt1_joint
    Vertex: a1_link, Edge: a1
      Vertex: a2_outer, Edge: a2_outer_joint
        Vertex: a2_inner_a2-a3_link, Edge: a2_inner_joint
          Vertex: a3-a3b_link, Edge: a3
            Vertex: a3b_drive, Edge: a3b_drive_joint
            Vertex: a3b_extension, Edge: a3b
              Vertex: toolbox, Edge: a3b_extension_toolbox_joint
                Vertex: a4-a5_link, Edge: a4
                  Vertex: a5-a6_link, Edge: a5
                    Vertex: a4_pin, Edge: a4_pin_joint
                    Vertex: a6_link, Edge: a6
                Vertex: a4_drive, Edge: a4_drive_joint
No non-tree joints.

julia> state = MechanismState(boom)
MechanismState{Float64, Float64, Float64, …}(…)

julia> set_configuration!(state, mast_a1_actuator_pt1_joint, 0.0)
ERROR: UndefVarError: mast_a1_actuator_pt1_joint not defined
```
Apologies if I am missing something obvious.

rdeits, 2018-10-08:
`parse_urdf` does not add any variables to your local scope. In fact, doing so would be literally impossible, since a function call cannot cause variables to be created in your local scope. To assign a joint to a local variable, the easiest thing to do would be:

```julia
joint = findjoint(mechanism, "mast_a1_actuator_pt1_joint")
set_configuration!(state, joint, 0.0)
```

togo59, 2018-10-08:
Got it. All the parameters are in the Mechanism, I just need to iterate over it to pull them out. Thank you. (BTW I am also following Duindam's thesis. I'm delighted that RBD is so close to his work.)

closed 2018-10-08 by togo59