JuliaRobotics/RigidBodyDynamics.jl #518: Cannot add RigidBodyDynamics

prajval10, 2018-11-21:
I have the latest version of `Julia` and `StaticArrays` but I am unable to add `RigidBodyDynamics`. The error is as follows:
```
(v1.0) pkg> add RigidBodyDynamics
 Resolving package versions...
ERROR: Unsatisfiable requirements detected for package StaticArrays [90137ffa]:
 StaticArrays [90137ffa] log:
 ├─possible versions are: [0.0.1-0.0.11, 0.1.0-0.1.5, 0.2.0-0.2.1, 0.3.0-0.3.1, 0.4.0, 0.5.0-0.5.1, 0.6.0-0.6.6, 0.7.0-0.7.2, 0.8.0-0.8.3, 0.9.0-0.9.2] or uninstalled
 ├─restricted to versions 0.9.2 by an explicit requirement, leaving only versions 0.9.2
 └─restricted by compatibility requirements with RigidBodyDynamics [366cf18f] to versions: [0.0.4-0.0.11, 0.1.0-0.1.5, 0.2.0-0.2.1, 0.3.0-0.3.1, 0.4.0, 0.5.0-0.5.1, 0.6.0-0.6.6, 0.7.0-0.7.2, 0.8.0-0.8.3] — no versions left
   └─RigidBodyDynamics [366cf18f] log:
     ├─possible versions are: [0.0.1-0.0.6, 0.1.0, 0.2.0, 0.3.0, 0.4.0, 0.5.0, 0.6.0-0.6.1, 0.7.0, 0.8.0, 0.9.0, 1.0.0-1.0.2, 1.1.0] oruninstalled
     └─restricted to versions * by an explicit requirement, leaving only versions [0.0.1-0.0.6, 0.1.0, 0.2.0, 0.3.0, 0.4.0, 0.5.0, 0.6.0-0.6.1, 0.7.0, 0.8.0, 0.9.0, 1.0.0-1.0.2, 1.1.0]
```

tkoolen, 2018-11-21:
I had to add upper bounds to StaticArrays because of #505, but in #515 I worked around that issue, so `add RigidBodyDynamics#master` should work now with the latest `StaticArrays` version. `add RigidBodyDynamics` will work again once the 1.2.0 version tag is merged (https://github.com/JuliaLang/METADATA.jl/pull/19657, PR opened yesterday).

tkoolen, 2018-11-23:
The 1.2.0 tag PR has been merged, so you should be able to do `add RigidBodyDynamics` again now (or `free RigidBodyDynamics` if you previously did `add RigidBodyDynamics#master`).

closed 2018-11-23 by tkoolen