JuliaRobotics/RigidBodyDynamics.jl #540: Support for SDF?

Nate711, 2019-02-04:
Hi, does RigidBodyDynamics.jl support extracting dynamics from SDF files?

Nate711, 2019-02-09:
Thanks for the info! Does the parser support any kind of URDF extension to model closed kinematic chains?

tkoolen, 2019-05-17:
I started exploring SDF support a bit, but I found that the behavior of SDF's `<frame>` tags is currently very ill-defined. I think I'm going to wait at least until a proposal to address this, https://bitbucket.org/osrf/sdf_tutorials/pull-requests/7/pose-frame-semantics-proposal-for-new/diff, has been merged; the proposed specification, http://sdformat.org/tutorials?branch=pose_frame_semantics_proposal&tut=pose_frame_semantics_proposal, looks a lot more sensible.