JuliaRobotics/RigidBodyDynamics.jl #546: Possible substitute for MuJoCo

kraftpunk97, 2019-04-06:
Hi. I discovered RigidBodyDynamics.jl thru a video on YouTube. I am very much interested in porting a RL gym to Julia and I was wondering if I can use RigidBodyDynamics.jl instead of MuJoCo for the robotics based environments. Thanks.

tkoolen, 2019-04-06:
What is your definition of a crash?

kraftpunk97, 2019-04-07:
@tkoolen Sorry if I'm unable to get my point across; I'm still trying to figure out what needs to be done. I am working on bringing the Lunar Lander arcade game to Julia which is written using the Box2D library. While searching for Julian alternatives for the library, I found RigidBodyDynamics.jl. But [your comment in this issue](https://github.com/JuliaRobotics/RigidBodyDynamics.jl/issues/110#issuecomment-257652238) says that "contact", (Sorry, I meant to say "contact" instead of "crash" in my original post) is not yet implemented. But that issue is from 2016, and I was wondering if this feature has been implemented. I had a quick glance over the docs, but I didn't find anything about it or I must have missed it in my hurry. Again, I apologise for being unclear.

kraftpunk97, 2019-04-08:
pinging @tkoolen

tkoolen, 2019-04-09:
Contact is currently limited to body-fixed contact points interacting with an environment consisting of halfspaces. Unfortunately I haven't had the time / personal need to focus on this beyond the current capabilities for a while now. I may need more complex geometry at some point in the coming months for my own work, but that's the status quo.

closed 2019-05-24 by kraftpunk97