JuliaRobotics/RigidBodyDynamics.jl #548: Issues using threading and StateCache's

blandry, 2019-05-09:
I am trying to parallelize some computation and I think I am running into issues from the implementation of the `StateCache`. I am not sure if this stems from the fact the `StateCache` is multi-threaded itself or from some underlying data structure in `mechanism` that is shared across caches but this self-contained example works fine when the loop is run serially, but fails when multi-threaded. If you inspect the cache in this example the "second" one doesn't end up being set. 

A `deepcopy` of mechanism before passing it to the `StateCache` constructor doesn't help.

Thanks!

```
using RigidBodyDynamics
using Base.Threads
using Distributed

N = 2

urdf = joinpath(dirname(pathof(RigidBodyDynamics)), "..", "test", "urdf", "Acrobot.urdf")
mechanism = parse_urdf(urdf)

sc = [StateCache(mechanism) for n = 1:N]
qs = rand(num_positions(mechanism),N)
vs = rand(num_velocities(mechanism),N)

# for n = 1:N
@threads for n = 1:N
# @distributed for n = 1:N
    state = sc[n][Float64]
    set_configuration!(state, qs[:,n])
    set_velocity!(state, vs[:,n])
    setdirty!(state)
end

# println(qs)
# for n = 1:N
#     println(configuration(sc[n][Float64]))
# end

for n = 1:N
    @assert configuration(sc[n][Float64]) == qs[:,n]
end

for n = 1:N
    @assert velocity(sc[n][Float64]) == vs[:,n]
end
```

tkoolen, 2019-05-10:
I can reproduce the assertion errors, but I'm not sure if this should be considered is a bug / usability issue. A documentation issue definitely.

`StateCache` actually takes care of threading internally already; the key it uses to index into its internal vector of `MechanismState`s is also based on the `threadid()`:

https://github.com/JuliaRobotics/RigidBodyDynamics.jl/blob/bb23110059c8b3ad94fc7fcc8dcd7d92514900d6/src/caches.jl#L7

So in the `@threads for` loop, each `state = sc[n][Float64]` makes it so that `sc[n]` creates a `MechanismState` that's associated with thread `n`. Then in the non-threaded `for` loops with the assertions, `sc[n]` is asked for the `MechanismState` that's associated with thread ID 1 (the main thread). So if `n != 1`, that actually creates a new `MechanismState` with zero `q` and `v`. So if you change the assertion `for` loops to `@threads for`, the tests pass. But you can also just use a single `StateCache`:

```julia
using RigidBodyDynamics
using Base.Threads
using Distributed

N = 2

urdf = joinpath(dirname(pathof(RigidBodyDynamics)), "..", "test", "urdf", "Acrobot.urdf")
mechanism = parse_urdf(urdf)

sc = StateCache(mechanism)
qs = rand(num_positions(mechanism),N)
vs = rand(num_velocities(mechanism),N)

@threads for n = 1:N
    state = sc[Float64]
    set_configuration!(state, qs[:,n])
    set_velocity!(state, vs[:,n])
    setdirty!(state)
end

@threads for n = 1:N
    @assert configuration(sc[Float64]) == qs[:,n]
end

@threads for n = 1:N
    @assert velocity(sc[Float64]) == vs[:,n]
end
```

And even though we had a single `StateCache`,

```julia
julia> sc.keys
2-element Array{Tuple{UInt64,Int64},1}:
 (0x042d6f95dce682b3, 2)
 (0x042d6f95dce682b3, 1)

julia> sc.values
2-element Array{MechanismState,1}:
 MechanismState{Float64, Float64, Float64, …}(…)
 MechanismState{Float64, Float64, Float64, …}(…)
```

tkoolen, 2019-05-10:
What do you think of this behavior? I might reconsider having the thread ID be part of the key.

blandry, 2019-05-10:
That makes sense thank you. I agree it's definitely not a bug.

I guess I tried to prevent race conditions myself by creating an array of caches so that each thread would have its own (each knot point along a trajectory actually). But this turned out to be a problem because I also split operations into two different threaded loops: one of them setting states and the other accessing two different state caches (think evaluation of a constraint on state i and i+1). I guess this becomes a problem if the threadid of the loop accessing the caches doesn't match the one that set it, which is only a problem to begin with because I tried to prevent race conditions on my side instead of trusting the cache to handle it.

I think this is fine behavior-wise. Maybe a note in the docs that "indexing combines threadid and type" at most if anything. But definitely feel like this was a mistake on my end.

I am switching to only one cache and that should be fine I think (I still need to fix other threading related issues in my contact jacobian cache to make this work).

tkoolen, 2019-05-10:
I can't find the original discussion @rdeits and I had on why to have the `threadid` as part of the key (it's not in https://github.com/JuliaRobotics/RigidBodyDynamics.jl/issues/363), but I believe we thought it was just *necessary* for `objectid(T)` to be valid across threads, not as a convenience. If that's the case, I think that stemmed from a misunderstanding of Julia's threading model at the time. 

@rdeits, if you agree with this assessment, I think we actually should change this. No other dictionary-like object in Julia behaves this way, and there might be cases where you *do* want multiple threads to access the same `MechanismState` from a `StateCache`. It's just a poor separation of concerns.

tkoolen, 2019-05-10:
@blandry, I don't know if the code in the description is representative of your actual code, but just as a comment: be aware of [false sharing](https://en.wikipedia.org/wiki/False_sharing) with stuff like

```julia
qs = rand(num_positions(mechanism),N)
```

Here, all of the `q`s in the columns will be contiguous in memory, and so there's a danger that the cache lines for the separate threads overlap, which can cause a significant performance decrease. So I think this would be bit better:

```julia
qs = Vector{Vector{Float64}}(undef, N)
@threads for i = 1 : N
    qs[i] = rand(num_positions(mechanism))
end
```

in accordance with [this post](https://discourse.julialang.org/t/poor-performance-on-cluster-multithreading/12248/17?u=tkoolen).

rdeits, 2019-05-10:
Yeah, that's fine. I chose to include the thread ID in the dictionary key because I assumed it might be useful to make the data structure thread-safe, but if it's not helping then we don't need to do it. I never had any reason to believe it was *required*, just that it might be helpful.

tkoolen, 2019-05-10:
> I never had any reason to believe it was required, just that it might be helpful.

OK, then I just misunderstood that at the time.

> it might be useful to make the data structure thread-safe

I don't think this is really a question of thread-safety then. Otherwise you could say that e.g. `Dict` isn't thread-safe either.

rdeits, 2019-05-10:
Sure, fair point. I guess I mean "thread-convenient" which, notably, is not a real thing :slightly_smiling_face:

blandry, 2019-05-11:
Yes I mean it definitely works fine and makes the code more minimal on my side. But even though it's obvious in the code I guess it isn't completely obvious from the API... and my reflex was to watch out for concurrency issues on my side of things with no expectation from the library.

@tkoolen thanks for your comment on the false sharing. I just wrote the snippet for this gh issue but it's good to know for the future!

Feel free to close this issue (I can also make a PR with the one line change this should require if that's the way you guys decide to go).

blandry, 2019-05-11:
Also thanks to both of you for being so responsive 👍

PR #549 (2019-05-13): Change multi-threading behavior of caches (AbstractTypeDicts).
Fix #548.

This is a breaking change, although I doubt that a lot of people have
been relying on this behavior up to this point.

Previously, `AbstractTypeDict` subtypes such as `StateCache`,
`DynamicsResultCache`, and `SegmentedVectorCache` would use the thread
ID for the thread in which the value (e.g. `MechanismState`) was created as part
of the key. This was not needed for thread-safety (a misunderstanding on
my part), and led to some unexpected behavior (#548) as it wasn't
documented properly. I also now consider it to be a po [... cut]
commit efa1a276 (2019-05-13): Change multi-threading behavior of caches (AbstractTypeDicts).

Fix #548.

This is a breaking change, although I doubt that a lot of people have
been relying on this behavior up to this point.

Previously, `AbstractTypeDict` subtypes such as `StateCache`,
`DynamicsResultCache`, and `SegmentedVectorCache` would use the thread
ID for the thread in which the value (e.g. `MechanismState`) was created as part
of the key. This was not needed for thread-safety (a misunderstanding on
my part), and led to some unexpected behavior (#548) as it wasn't
documented properly. I also now consider it to be a p [... cut]
closed 2019-05-13 by tkoolen