JuliaRobotics/RigidBodyDynamics.jl #559: Do you support damping in urdf?

tkoolen, 2019-06-20:
Do you support damping in urdf? 
`<dynamics damping="0.6"/>`
If not, how can I add one?

_Originally posted by @Islam0mar in https://github.com/JuliaRobotics/RigidBodyDynamics.jl/issues/317#issuecomment-503895047_

tkoolen, 2019-06-20:
Opened a new issue for this.

Damping coefficients are not stored in RigidBodyDynamics `Joint`s; in general the `dynamics` tag for a joint is ignored. The motivation for that is that the friction model in the URDF spec is just one model that I didn't want to get tied down to.

I've used the following code in the past to parse damping values from the URDF *after creating a `Mechanism`* (by parsing the same URDF).

https://github.com/tkoolen/QPWalkingControl.jl/blob/master/src/damping.jl

The resulting `JointDamping` object is callable, and can be used as the `control!` function with RigidBodySim (possibly combined with other controllers using a `SumController`). See the cell with comment `# create ODEProblem` in https://github.com/tkoolen/QPWalkingControl.jl/blob/master/notebooks/Atlas%20walking.ipynb. If you don't want to use RigidBodySim.jl, you can can use the same `JointDamping` callable object with `RigidBodyDynamics.simulate`.

I wouldn't actually add a dependency on QPWalkingControl to get this functionality; just copy the file to your own project for now. I think I'll move it to RigidBodyDynamics at some point.

Islam0mar, 2019-06-20:
Thanks, I will try that.