JuliaRobotics/RigidBodyDynamics.jl #565: A simple hello world that involves moving a box across a straight line.

kraftpunk97, 2019-07-31:
Hello. I studied the double pendulum example but I'm trouble understanding it, and I wanted to try something even simpler. I want to start by declaring a box and making it move a certain distance and simulate that. Is there an example detailing that or something similar?

kraftpunk97, 2019-08-04:
pinging @tkoolen...

tkoolen, 2019-08-05:
OK, here's an even simpler example, tracking a sinusoidal reference trajectory for a single body moving along the y-axis, simulated using `RigidBodyDynamics.simulate` and animated using `MeshCatMechanisms` (showing the inertial ellipsoid). You'll probably want to use RigidBodySim, but I think the RigidBodySim quickstart notebook should allow you to readily adapt this example.

```julia
using RigidBodyDynamics, MeshCatMechanisms
using RigidBodyDynamics.PDControl
using LinearAlgebra

world = RigidBody{Float64}("world")
mechanism = Mechanism(world)
bodyframe = CartesianFrame3D("body")
inertia = SpatialInertia(bodyframe,
    moment_about_com=Diagonal([0.1, 0.1, 0.1]),
    com=zeros(3),
    mass=1.0)
body = RigidBody(inertia)
joint = Joint("joint", Prismatic([0.0, 1.0, 0.0]))
attach!(mechanism, world, body, joint)

visualizer = MechanismVisualizer(mechanism)
open(visualizer)

control! = let joint=joint
    function(τ::SegmentedVector, t::Number, state::MechanismState)
        gains = PDGains(100.0, 20.0)
        q = configuration(state, joint)
        v = velocity(state, joint)
        qref = sin(t)
        vref = cos(t)
        τjoint = τ[joint]
        τjoint .= pd.(Ref(gains), q, qref, v, vref)
    end
end

state = MechanismState(mechanism)
ts, qs, vs = simulate(state, 10.0, control!, Δt=1e-1)
animate(visualizer, ts, qs)
```