JuliaRobotics/RigidBodyDynamics.jl #566: Support for spring elements

touste, 2019-08-30:
Are spring connections supported? If not, that would be a nice addition.
Their action would simply result in a force proportional to the relative displacement of two points.
Thanks!

tkoolen, 2019-08-31:
Yep, that'd be a nice addition. It's not that hard to implement currently, through the use of the `external_wrenches` argument of `dynamics!` in combination with the `Wrench` constructor that takes an application point and a force, though `simulate` currently doesn't expose the `external_wrenches` API to the controller. But it would be nice to just have a `LinearSpring` type or something like that.

Right now I'm almost exclusively focused on my thesis and thesis defense (until September 30 or so). After that I plan to give this package some more love for a bit.

touste, 2019-09-04:
Good to know this is of interest! All the best for your thesis defense!

touste, 2019-10-14:
Hi, I hope your PhD defense went smoothly!
I'd like to try implementing the spring connectors in this package, however I find it difficult to understand the way I should do that. I'm more used to working with solid mechanics and not really familiar with rigid body dynamics, could you point me to how I could implement this? I expect a link type for which are given forces and positions as input, and it should output new positions and perhaps derivatives, but I cannot find it as such.
Thanks!

lieskjur, 2021-10-21:
Hi, late to the party here. I am implementing compliant joints for a project and as a proof of concept I defined a "spring" joint type which is practically a prismatic joint with added parameters `free_length` and `stiffness`. The main difference is that inside `dynamics!` it produces a joint torque that is added to `torque` provided as an argument.