JuliaRobotics/RigidBodyDynamics.jl #568: NaN in simulation

kuro26, 2019-09-05:
this is the source code:
`
using RigidBodyDynamics
using Rotations
urdf = "singleBox.urdf"
mechanism = parse_urdf(urdf, floating=true) 

box = bodies(mechanism)[2]   
Nmdl = RigidBodyDynamics.Contact.hunt_crossley_hertz()   
Fmdl = RigidBodyDynamics.Contact.ViscoelasticCoulombModel(0.3, 30e3,0.3)   
scm = RigidBodyDynamics.Contact.SoftContactModel(Nmdl, Fmdl) 
for i in [-0.1, 0.1]
    for j in [-0.1, 0.1]
        for k in [-0.1, 0.1]
            point = Point3D(default_frame(box), i, j, k)
            cp = RigidBodyDynamics.Contact.ContactPoint(point, scm)
            add_contact_point!(bodies(mechanism)[2], cp)
        end
    end
end
frame = default_frame(bodies(mechanism)[1])
point = Point3D(frame, 0.0, 0.0, 0.0)
normal = FreeVector3D(frame, 0., 0., 1.)
halfspace = RigidBodyDynamics.Contact.HalfSpace3D(point, normal)

push!(mechanism.environment.halfspaces, halfspace)

function no_control!(torques::AbstractVector, t, state::MechanismState)
    for i in 1:length(torques)
        torques[i] = 0
    end
end
floating_joint = joints(mechanism)[1]
state = MechanismState(mechanism)
zero_velocity!(state)
rot = RotXYZ{Float64}(0.2, 0.4, 0.6)
quat = convert(Quat{Float64}, rot)
set_configuration!(state, floating_joint, [quat.w, quat.x, quat.y, quat.z, 0.5, 0.3, 2])

final_time = 5
ts, qs, vs = simulate(state, final_time, no_control!; Δt = 1e-4);

using MeshCatMechanisms
mvis = MechanismVisualizer(mechanism, URDFVisuals(urdf));
open(mvis)
MeshCatMechanisms.animate(mvis, ts, qs; realtimerate = 1.);
`
the urdf file:
`
<?xml version="1.0"?>
<robot name="singleBox">
    <link name = "box">
        <visual>
            <geometry>
                <box size="0.2 0.2 0.2"/>
            </geometry>
            <material name="red">
                <color rgba="1.0 0.0 0.0 0.7"/>
            </material>
        </visual>
        <collision>
            <geometry>
                <box size="0.2 0.2 0.2"/>
            </geometry>
        </collision>
        <inertial>
            <mass value="1.0"/>
            <inertia ixx="0.2" ixy="0.0" ixz="0.0" iyy="0.2" iyz="0.0" izz="0.2"/>
        </inertial>
    </link>
</robot>
`
problem is:
NaN in simulation result after 1.8 seconds.

tkoolen, 2019-09-05:
I can probably take a look this weekend. Thanks for the reproducible example and clear issue description. I suspect that the large spring stiffness compared to the inertial parameters of the box results in integration instability with a step size of `1e-4`; you may need to tune some of these parameters, and/or switch to using RigidBodySim with a variable time step solver like `Tsit5` from DifferentialEquations.jl.

kuro26, 2019-09-06:
Thank you for your concern, parameter tunning seems doesn't solve, I might need to look into the integrator.

tkoolen, 2019-09-06:
I just tried with `Δt = 1e-5`, and this appears to fix the `NaN` issue. Could it be that you continued the simulation from a `NaN` state (didn't reset it to the same initial state) after you changed the `Δt`?

tkoolen, 2019-09-06:
But you'll probably want to pick a lower value of `k` for the `HuntCrossleyModel` (normal force model) and `ViscoelasticCoulombModel` (tangential force model), for such a light box, so that you can simulate faster.

kuro26, 2019-09-07:
Thank you very much, I didn't try 1e-5 before, seems I can still use your integrator! This library is really easy to use, I wish you can keep maintaining it to make it better!

closed 2019-09-09 by kuro26