JuliaRobotics/RigidBodyDynamics.jl #572: Support for Multiple Robots (Mechanisms?)

ericjang, 2019-09-29:
Hello,

Great work with this suite of packages! I came across this yesterday and have been playing around with the notebooks to learn more about rigid body dynamics. It appears that RigidBodyDynamics.jl and related libraries are designed around control of single robots (which maps to a single "Mechanism" data structure). For instance, MeshCatVisualizer, Dynamics, solver interfaces, etc.

I am wondering how to design a scene with the following scenarios:
1. A set of objects disconnected from the robot mechanism. For instance, a grasping an manipulation task.
2. Multiple robot Mechanisms (e.g. to have Atlas and Valkyrie play soccer against each other :) ). There should be logic to handle ODE solutions separately when these are not in contact, and I was wondering if there was anything on the development roadmap for this.

tkoolen, 2019-10-05:
Sorry about the response time. I'll get to this on Sunday.

tkoolen, 2019-10-07:
So one thing you can do already is create a `Mechanism` with just a root body (fixed world), and then `attach!` various different mechanisms (perhaps each loaded from a URDF) to that root body. That way all of the simulation tools in this package and in RigidBodySim still just work, essentially with the configuration and velocity vectors concatenated. However, the `URDFVisuals` from MechanismGeometries won't work so well with this merged `Mechanism`, and I recognize that it might be preferable anyway to keep the `Mechanism`s separate.

I was actually working on support for simulation with multiple objects that provide state in the context of a reimplementation of the contact code, so this feature might just be a byproduct of that work. I'll keep your comment in mind as I (hopefully) wrap that up over the course of the next week.

(On the 3D visualization side, MeshCat can handle multiple robots through its own tree structure, where you could have a `vis[:atlas]` sub-visualizer and a `vis[:valkyrie]` subvisualizer.)