JuliaRobotics/RigidBodyDynamics.jl #581: URDF.parse and LightXML fail during precompilation

Crown421, 2020-02-04:
I have been using RigidBodyDynamics as part of a module for one of my project. However, I have now twice run into the same error message when precompiling my module. 
```
I/O warning : failed to load external entity "../rs_alt.urdf"
ERROR: LoadError: LoadError: LightXML.XMLParseError{String}("Failure in parsing an XML file.")
Stacktrace:
 [1] _check_result at /opt/.julia/packages/LightXML/W8FVT/src/document.jl:88 [inlined]
 [2] parse_file(::String) at /opt/.julia/packages/LightXML/W8FVT/src/document.jl:92
 [3] #parse_urdf#5(::Type{Float64}, ::Bool, ::Dict{String,UnionAll}, ::RigidBodyDynamics.Fixed{Float64}, ::Bool, ::StaticArrays.SArray{Tuple{3},Float64,1,3}, ::Nothing, ::Nothing, ::typeof(RigidBodyDynamics.URDF.parse_urdf), ::String) at /opt/.julia/packages/RigidBodyDynamics/2qxSc/src/urdf/parse.jl:183
 [4] (::RigidBodyDynamics.URDF.var"#kw##parse_urdf")(::NamedTuple{(:scalar_type, :remove_fixed_tree_joints),Tuple{DataType,Bool}}, ::typeof(RigidBodyDynamics.URDF.parse_urdf), ::String) at ./none:0
 [5] parse_urdf(::Type, ::String) at /opt/.julia/packages/RigidBodyDynamics/2qxSc/src/urdf/parse.jl:236
 [6] top-level scope at /opt/.julia/packages/FK/ZTEcO/src/rskinematics.jl:4
 [7] include at ./boot.jl:328 [inlined]
 [8] include_relative(::Module, ::String) at ./loading.jl:1105
 [9] include at ./Base.jl:31 [inlined]
 [10] include(::String) at /opt/.julia/packages/FK/ZTEcO/src/FK.jl:7
 [11] top-level scope at /opt/.julia/packages/FK/ZTEcO/src/FK.jl:21
 [12] include at ./boot.jl:328 [inlined]
 [13] include_relative(::Module, ::String) at ./loading.jl:1105
 [14] include(::Module, ::String) at ./Base.jl:31
 [15] top-level scope at none:2
 [16] eval at ./boot.jl:330 [inlined]
 [17] eval(::Expr) at ./client.jl:425
 [18] top-level scope at ./none:3
```

Unfortunately, I can not share the urdf in question (or the module), but to add a few more details. Working with this has been fine on my Linux machine, but when I moved everything to my Mac the error appeared for the first time. Strangely, explicitly adding RigidBodyDynamics in the REPL, and running the same URDF.parse command worked, and it has continued to work since. 

Now I have run into the same error trying to install my package into a Singularity container. Once again, `shell`ing into the container and running the command explicitly worked once, but now I get this error after `using RigidBodyDynamics`
```
[ Info: Precompiling RigidBodyDynamics [366cf18f-59d5-5db9-a4de-86a9f6786172]
ERROR: LoadError: KeyError: key Rotations [6038ab10-8711-5258-84ad-4b1120ba62dc] not found
Stacktrace:
 [1] getindex at ./dict.jl:477 [inlined]
 [2] root_module at ./loading.jl:962 [inlined]
 [3] require(::Base.PkgId) at ./loading.jl:928
 [4] require(::Module, ::Symbol) at ./loading.jl:917
 [5] include at ./boot.jl:328 [inlined]
 [6] include_relative(::Module, ::String) at ./loading.jl:1105
 [7] include(::Module, ::String) at ./Base.jl:31
 [8] top-level scope at none:2
 [9] eval at ./boot.jl:330 [inlined]
 [10] eval(::Expr) at ./client.jl:425
 [11] top-level scope at ./none:3
in expression starting at /opt/.julia/packages/RigidBodyDynamics/2qxSc/src/RigidBodyDynamics.jl:6
ERROR: Failed to precompile RigidBodyDynamics [366cf18f-59d5-5db9-a4de-86a9f6786172] to /home/steffen/OneDrive/Code/JPLCode/Code/FKaux/singularity/precompile/compiled/v1.3/RigidBodyDynamics/WeevQ_5i4om.ji.
Stacktrace:
 [1] error(::String) at ./error.jl:33
 [2] compilecache(::Base.PkgId, ::String) at ./loading.jl:1283
 [3] _require(::Base.PkgId) at ./loading.jl:1024
 [4] require(::Base.PkgId) at ./loading.jl:922
 [5] require(::Module, ::Symbol) at ./loading.jl:917
```

I will probably fix the issue by removing RigidBodyDynamics as an explicit dependency as its only needed once, but I thought I would open an issue anyway. 

Happy to provide further information as far as I can.

tkoolen, 2020-02-05:
First, are you sure `"../rs_alt.urdf"` exists? I've seen this error when you're trying to load a file that doesn't actually exist. If that's not it, can you reduce the urdf until the problem goes away. Can you reduce it until you have something you can share? What's special about this particular urdf that could cause the problem?

rdeits, 2020-02-05:
The IO error seems to indicate that you're loading your URDF file *during
precompilation*. That's a pretty odd thing to be doing, and it might not
work the way you expect. Can you move the URDF load to your `__init__()`
function instead?

Crown421, 2020-02-05:
It is part of my module code, as I effectively extract some transformation matrices from it, which are then put into a `const`, which I am hoping allows the compiler to to inline the specific values into the function code. I might be wrong here though. 

More specifically, the code works on most machines, and the file exists at the indicated path, relative to the file containing the command. However, after switching to some `@__DIR__` construct to use effectively the absolute path it works everywhere.

Therefore, it is likely some problem unrelated to this package.

closed 2020-02-05 by Crown421