JuliaRobotics/RigidBodyDynamics.jl #584: Updating joint transforms after creating a mechanism

Nate711, 2020-04-20:
Is it possible to update the frame transformations of a joint after using the attach! function to insert it into the mechanism? I was attempting to change the joint transform by modifying joint.joint_successor[] but at least what I did didn't work. I'm guessing something happens with attach! that triggers caching or some other pre-calculation that doesn't get updated when I modify the joint transform directly. Thanks!

tkoolen, 2020-04-22:
Yes, should be possible, using `set_joint_to_predecessor` (undocumented), or by directly setting `joint.joint_successor[]` as you did. If you created a `MechanismState` for your `Mechanism` and you didn't do anything else to cause the cache to be invalidated, you'll have to manually call `setdirty!(state)` though.

Nate711, 2020-04-22:
Would there be anything particular to modifying non-tree joints? It seems like I can easily modify tree joints, but when I try it for non-tree joints (which I want in the end), it doesn't work so well.

Nate711, 2020-04-23:
Hey Twan, wanted to follow up and provide some code in the Four Bar Linkage example that shows the issue. 
1) Modifying the non-tree joint, joint 4, with this code:
```joint4.joint_to_successor[] = Transform3D(joint4.joint_to_successor[].from, joint4.joint_to_successor[].to, SVector(l_3*0.5,0,0))```
after attaching joint4 results in a strange abnormality between bodies 2 and 3. 
[image]

2) While modifying the tree-joint joint3 after it's attached with this code:
```joint3.joint_to_predecessor[] = Transform3D(joint3.joint_to_predecessor[].from, joint3.joint_to_predecessor[].to, SVector(l_0*0.5,0,0))```
results in a working mechanism.
[image]

In both cases ```setdirty!(state)``` was called before simulating and after the mechanism was modified.

Nate711, 2020-04-28:
@tkoolen just wanted to bump the issue. Thanks!