JuliaRobotics/RigidBodyDynamics.jl #589: Body wrench from forces applied at different points

ferrolho, 2020-07-24:
Hello, @tkoolen! :slightly_smiling_face: I am trying to model the surface-to-surface contact between a humanoid's foot and flat ground. For that, I am defining four contact points per foot, similar to what you did for [Atlas](https://github.com/tkoolen/AtlasRobot.jl/blob/c5f54824c2d13b40097056f0b5b2fe7da9f3eeeb/src/AtlasRobot.jl#L34-L44) and [Valkyrie](https://github.com/tkoolen/ValkyrieRobot.jl/blob/2e352edffad0d17267681fef9f8ec59f615a4763/src/ValkyrieRobot.jl#L63-L66). The image below shows the four contact points of the left foot, as well as a triad located at the frame origin of the sole.

[image]

This next image shows some forces being applied at each contact point.

[image]

My question is: for each foot, how can I compute the total wrench acting on the link given the individual contact forces? I want to do this so I can call `dynamics!(result, state, torques, externalwrenches)` with the external wrenches acting on each foot.

I will leave the code for a single force applied at the frame origin below, as it may be a helpful reference for the discussion:

```julia
frame_foot = findbody(mechanism, "left_sole_link")

force_in_world_frame = FreeVector3D(root_frame(mechanism), [0.0, 0.0, 100.0])
force_in_foot_frame = transform(state, force_in_world_frame, frame_foot)

wrench = Wrench(Point3D(frame_foot, zeros(T, 3)), force_in_foot_frame)

externalwrenches = Dict{BodyID,Wrench{T}}(
    BodyID(findbody(mechanism, "leg_left_6_link")) => transform(state, wrench, root_frame(mechanism)),
)

result = robot.dynamicsresultcache[T]
dynamics!(result, state, torques, externalwrenches)
```

ferrolho, 2020-07-24:
Well, this is embarrassing... I was really stuck with the assumption that the first argument I was passing to `Wrench()` was the angular component... It turns out I was missing that there is a `Wrench()` that is dispatched on a `Point3D` for wrenches defined as a force applied to an application point. (Even though I have this explicitly written in my snippet above...)

[image]

Oh well, correct me if I am wrong, but what I was looking for is therefore something like this:

```julia
frame_foot = default_frame(findbody(mechanism, "left_sole_link"))

force_1 = transform(state, FreeVector3D(root_frame(mechanism), [0.0, 0.0,  20.0]), frame_foot)
force_2 = transform(state, FreeVector3D(root_frame(mechanism), [0.0, 0.0,  20.0]), frame_foot)
force_3 = transform(state, FreeVector3D(root_frame(mechanism), [0.0, 0.0, 100.0]), frame_foot)
force_4 = transform(state, FreeVector3D(root_frame(mechanism), [0.0, 0.0, 100.0]), frame_foot)

wrench_1 = Wrench(Point3D(frame_foot, [ 0.1,  0.05, 0.0]), force_in_foot_frame)
wrench_2 = Wrench(Point3D(frame_foot, [ 0.1, -0.05, 0.0]), force_in_foot_frame)
wrench_3 = Wrench(Point3D(frame_foot, [-0.1,  0.05, 0.0]), force_in_foot_frame)
wrench_4 = Wrench(Point3D(frame_foot, [-0.1, -0.05, 0.0]), force_in_foot_frame)

wrench = wrench_1 + wrench_2 + wrench_3 + wrench_4

externalwrenches = Dict{BodyID,Wrench{T}}(
    BodyID(findbody(mechanism, "leg_left_6_link")) => transform(state, wrench, root_frame(mechanism)),
)

result = robot.dynamicsresultcache[T]
dynamics!(result, state, torques, externalwrenches)
```

closed 2020-07-25 by ferrolho

tkoolen, 2020-07-28:
Yep, that's right!

ferrolho, 2020-07-28:
[image]
:slightly_smiling_face: