JuliaRobotics/RigidBodyDynamics.jl #615: Non-allocating construction of `externalwrenches`

ferrolho, 2021-03-06:
Hi, Twan! :smiley: I have a question about the `externalwrenches` argument in the `dynamics!` and `inverse_dynamics!` methods. Consider the following callback:

```julia
function forward_dynamics_defects!(defects, robot, x::AbstractArray{T}, h) where {T}
    result = robot.dynamicsresultcache[T]
    state = robot.statecache[T]

    # Indices of the decision variables in argument `x`.
    ind_qᵢ = (1:robot.n_q)
    ind_q̇ᵢ = (1:robot.n_v) .+ (robot.n_q)
    ind_τᵢ = (1:robot.n_τ) .+ (robot.n_q + robot.n_v)
    ind_λᵢ = (1:robot.n_λ) .+ (robot.n_q + robot.n_v + robot.n_τ)
    ind_qq̇ᵢ₊₁ = (1 : robot.n_q + robot.n_v) .+ (robot.n_q + robot.n_v + robot.n_τ + robot.n_λ)

    # Place the decision variables in `SVector`s
    qᵢ = SVector{24}(@view x[ind_qᵢ])
    vᵢ = SVector{24}(@view x[ind_q̇ᵢ])
    τᵢ = SVector{18}(@view x[ind_τᵢ])
    λᵢ = SVector{15}(@view x[ind_λᵢ])

    set_configuration!(state, qᵢ)
    set_velocity!(state, vᵢ)

    # ==== !!! <START> !!! ====

    wrenches = [begin
        cp = robot.contact_points[j]
        ind_λᵢⱼ = (1:3) .+ (j - 1) * 3
        force = transform(state, FreeVector3D(root_frame(robot.mechanism), @view λᵢ[ind_λᵢⱼ]), cp.frame)
        wrench = Wrench(cp, force)
    end for j = 1:length(robot.contact_points)]

    externalwrenches = Dict{BodyID,Wrench{T}}(
        BodyID(body) => transform(state, sum(@view wrenches[cp_inds]), root_frame(robot.mechanism))
        for (body, cp_inds) ∈ robot.feet_bodies_to_cp_inds
    )

    # ===== !!! <END> !!! =====

    torques = @MVector zeros(T, 24)
    torques[1+num_velocities(robot.floating_joint):end] .= τᵢ
    torques = SVector(torques)

    dynamics!(result, state, torques, externalwrenches)
    v̇ᵢ = SVector{24}(result.v̇)

    next_state_integrated = @MVector zeros(T, 48)
    semi_implicit_euler_step!(next_state_integrated, robot, qᵢ, vᵢ, v̇ᵢ, h)
    next_state_integrated = SVector(next_state_integrated)

    next_state_discretised = @view x[ind_qq̇ᵢ₊₁]

    @. defects = next_state_integrated - next_state_discretised

    return defects
end
```

In the snippet above, I have tagged the `<START>` and the `<END>` of where I build a vector of wrenches (one per contact point) and an `externalwrenches` dictionary. That segment of code allocates quite a bit, and I believe it may be slowing down the NLP solver which calls this method to compute dynamics defects (in the context of trajectory optimisation).

Is there a better way to do this? I am wondering whether a new cache type (similar to the existing `StateCache` and `DynamicsResultCache`) would solve this. I'd really appreciate any pointers you can give me on this. Thank you very much in advance!

tkoolen, 2021-03-08:
Oh sure, you can avoid the allocations by preallocating the `Dict{BodyID,Wrench{T}}` outside of the function and passing it in as an argument; then you can just use `empty!` on the dict, followed by a series of `external_wrenches[body_id] = wrench_for_body` calls.

tkoolen, 2021-03-08:
Oh I guess you intend for this to be autodiffable. You can either pattern match e.g. `DynamicsResultCache` in `caches.jl` (inherit from `AbstractTypeDict`, implement `valuetype` and `makevalue`; it's really not that magical), or (a little hackier as it makes an assumption about the `dynamics!` implementation, but it works) you could actually just use `result.totalwrenches` as `externalwrenches`.

ferrolho, 2021-03-08:
Thanks for your reply! Yes, I intend to use ForwardDiff.jl on this; so the first approach wouldn't work, unfortunately.

I guess there's no avoiding the allocations from creating the wrenches, right? I mean, in
```julia
wrenches = [begin
    cp = robot.contact_points[j]
    ind_λᵢⱼ = (1:3) .+ (j - 1) * 3
    force = transform(state, FreeVector3D(root_frame(robot.mechanism), @view λᵢ[ind_λᵢⱼ]), cp.frame)
    wrench = Wrench(cp, force)
end for j = 1:length(robot.contact_points)]
```
because of
```julia
julia> let T = Ref(Wrench{Float64}), frame = Ref(root_frame(robot.mechanism))
           @btime rand($T[], $frame[])
       end
  182.560 ns (2 allocations: 80 bytes)
```

ferrolho, 2021-03-09:
If I create a new `struct` to be used in the cache's values, e.g., `MyWrenches`, it seems to work. I.e.,
```julia
mutable struct MyWrenches{T, M}
    mechanism::Mechanism{M}
    wrenches::BodyDict{Wrench{T}}

    function MyWrenches{T}(mechanism::Mechanism{M}) where {T, M}
        rootframe = root_frame(mechanism)
        wrenches = BodyDict{Wrench{T}}(b => zero(Wrench{T}, rootframe) for b in bodies(mechanism))

        new{T, M}(mechanism, wrenches)
    end
end

MyWrenches(mechanism::Mechanism{M}) where {M} = MyWrenches{M}(mechanism)

struct WrenchesCache{M} <: AbstractTypeDict
    mechanism::Mechanism{M}
    keys::Vector{UInt}
    values::Vector{MyWrenches{<:Any, M}}
end

Base.show(io::IO, ::WrenchesCache{M}) where {M} = print(io, "WrenchesCache{$M}(…)")

WrenchesCache(mechanism::Mechanism{M}) where {M} = WrenchesCache{M}(mechanism, [], [])
@inline valuetype(::Type{WrenchesCache{M}}, ::Type{T}) where {M, T} = MyWrenches{T, M}
@inline makevalue(c::WrenchesCache, ::Type{T}) where {T} = MyWrenches{T}(c.mechanism)
```
But is that really necessary?

ferrolho, 2021-03-10:
I've made some progress. :stuck_out_tongue: I updated the snippet regarding the `externalwrenches` to use the new `WrenchesCache`:
```julia
# Building `wrenches` takes: 2.289 μs (3 allocations: 816 bytes)
wrenches = Wrench{T}[]
for j = 1:length(robot.contact_points)
    cp = robot.contact_points[j]
    ind_λᵢⱼ = (1:3) .+ (j - 1) * 3
    force = transform(state, FreeVector3D(root_frame(robot.mechanism), @view λᵢ[ind_λᵢⱼ]), cp.frame)
    wrench = Wrench(cp, force)
    push!(wrenches, wrench)
end

# Building `externalwrenches` takes: 941 ns (0 allocations: 0 bytes)
externalwrenches = robot.wrenchescache[T].wrenches
for (body, cp_inds) ∈ robot.feet_bodies_to_cp_inds
    externalwrenches[body] = transform(state, sum(@view wrenches[cp_inds]), root_frame(robot.mechanism))
end
```

So far, the overall improvement for the `forward_dynamics_defects!` function is:
```julia
14.863 μs (75 allocations: 10.42 KiB)  # Before
10.870 μs ( 4 allocations: 880 bytes)  # Now
```

I guess that the last things we could shave off are those allocations from building the `wrenches` vector on the spot, right? Would that be solved by pre-allocating the `wrenches` vector in, e.g., `MyWrenches`?

tkoolen, 2021-03-12:
Why are you first allocating a new vector for `wrenches` and then populating `externalwrenches`? Can't you just do that in a single loop?

tkoolen, 2021-03-12:
Something like this (untested):

```julia
externalwrenches = robot.wrenchescache[T].wrenches # no need to add zero wrench entries for all bodies
empty!(externalwrenches)
for (body, cp_inds) ∈ robot.feet_bodies_to_cp_inds
    externalwrenches[body] = zero(Wrench{T}, root_frame(mechanism))
    # optimization assuming that cp.frame is the body frame for all contact points attached to this body:
    body_to_root = transform_to_root(state, body) 
    for cp_ind ∈ cp_inds
        cp_body = robot.contact_points[cp_ind]
        cp_root = transform(cp_body, body_to_root)
        ind_λᵢⱼ = (1:3) .+ (j - 1) * 3
        force_root = FreeVector3D(root_frame(robot.mechanism), @view λᵢ[ind_λᵢⱼ])
        externalwrenches[body] += Wrench(cp_root, force_root)
    end
end
```

ferrolho, 2021-03-12:
Yes, simplifying this to a single loop is something I can definitely do. Sorry, I was "stuck" on my approach.

1. The assumption that `cp.frame` corresponds to `body`'s frame is not entirely valid due to `remove_fixed_tree_joints!`. For example, when I load the mechanism for ANYmal, I add contact points to `"*_FOOT"`s which get deleted after `remove_fixed_tree_joints!`; so, afterwards, I need to handle them from their parents `"*_SHANK"`. The rationale of your optimisation in the snippet above still holds, I just need to do some bookkeeping to retrieve the correct bodies for the transformations. But for now, and to get back to the main discussion, I have stopped calling `remove_fixed_tree_joints!`.

2. I am still getting some allocations, even with the single loop: `(5 allocations: 560 bytes)`. They seem to be coming from `Wrench(cp_root, force_root)`, and they match the number of contact points I defined (five). I am trying to find out where those allocations are coming from. You would suppose your loop would be non-allocating, right?

3. I get an error calling `empty!(externalwrenches)`:
    ```julia
    MethodError: no method matching empty!(::RigidBodyDynamics.CustomCollections.IndexDict{BodyID, Base.OneTo{BodyID}, Wrench{Float64}})
    Closest candidates are:
      empty!(::Union{DataStructures.SortedDict, DataStructures.SortedMultiDict, DataStructures.SortedSet}) at /home/henrique/.julia/packages/DataStructures/ixwFs/src/container_loops.jl:321
      empty!(::MathOptInterface.Utilities.DoubleDicts.AbstractWithType{F, S, V, DI, DD} where {V, DI, DD}) where {F, S} at /home/henrique/.julia/packages/MathOptInterface/5WwpK/src/Utilities/DoubleDicts.jl:500
      empty!(::DataStructures.DefaultDict, ::Any...) at /home/henrique/.julia/packages/DataStructures/ixwFs/src/delegate.jl:37
      ...
    ```

4. As for your comment, `# no need to add zero wrench entries for all bodies`, are you referring to the constructor of `MyWrenches`? I tried to create an empty dictionary, `BodyDict{Wrench{T}}()`, instead of `BodyDict{Wrench{T}}(b => zero(Wrench{T}, rootframe) for b in bodies(mechanism))`; but if I do that, simply calling `robot.wrenchescache[Float64].wrenches` will trigger the following error:
    ```julia
    MethodError: no method matching RigidBodyDynamics.CustomCollections.IndexDict{BodyID, Base.OneTo{BodyID}, Wrench{Float64}}()
    Closest candidates are:
      RigidBodyDynamics.CustomCollections.IndexDict{K, KeyRange, V}(::Array{Pair{K, V}, 1}) where {K, KeyRange<:AbstractUnitRange{K}, V} at /home/henrique/.julia/dev/RigidBodyDynamics/src/custom_collections/IndexDict.jl:65
      RigidBodyDynamics.CustomCollections.IndexDict{K, KeyRange, V}(::KeyRange) where {K, KeyRange<:AbstractUnitRange{K}, V} at /home/henrique/.julia/dev/RigidBodyDynamics/src/custom_collections/IndexDict.jl:61
      RigidBodyDynamics.CustomCollections.IndexDict{K, KeyRange, V}(::Any) where {K, KeyRange<:AbstractUnitRange{K}, V} at /home/henrique/.julia/dev/RigidBodyDynamics/src/custom_collections/IndexDict.jl:86
      ...
    
    Stacktrace:
     [1] (RigidBodyDynamics.MyWrenches{Float64, M} where M)(mechanism::Mechanism{Float64})
       @ RigidBodyDynamics ~/.julia/dev/RigidBodyDynamics/src/caches.jl:129
     [2] makevalue
       @ ~/.julia/dev/RigidBodyDynamics/src/caches.jl:154 [inlined]
     [3] getindex(c::WrenchesCache{Float64}, #unused#::Type{Float64})
       @ RigidBodyDynamics ~/.julia/dev/RigidBodyDynamics/src/caches.jl:13
     [4] top-level scope
       @ In[23]:1
     [5] eval
       @ ./boot.jl:360 [inlined]
     [6] include_string(mapexpr::typeof(REPL.softscope), mod::Module, code::String, filename::String)
       @ Base ./loading.jl:1090
    ```

tkoolen, 2021-03-12:
>     3\. I get an error calling `empty!(externalwrenches)`:

> 4\. As for your comment, `# no need to add zero wrench entries for all bodies`, are you referring to the constructor of `MyWrenches`? I tried to create an empty dictionary, `BodyDict{Wrench{T}}()`, [...]

Oh right, you're using a `BodyDict` / `IndexDict`, which is my custom type of associative container that's really just a fancy vector. You can either fill with zeros as you were, or switch to a regular `Dict{BodyID, Wrench{T}}`. I don't think it'll matter too much in terms of performance in this case, actually, but you can benchmark.

>     I just need to do some bookkeeping to retrieve the correct bodies for the transformations. But for now, and to get back to the main discussion, I have stopped calling `remove_fixed_tree_joints!`.

Yeah, you should be able to just use the `fixed_transform` function to get the transform from whatever frame you defined the contact point in to the body's default body-fixed frame. Things are just easier at runtime if you transform all of the points attached to a body to a common frame beforehand.

> 2\. I am still getting some allocations, even with the single loop: `(5 allocations: 560 bytes)`. They seem to be coming from `Wrench(cp_root, force_root)`, and they match the number of contact [...]

How are you measuring allocations?

ferrolho, 2021-03-15:
> You can (...) switch to a regular `Dict{BodyID, Wrench{T}}`.

Yes, if I switch to the regular dictionary, then `empty!` works as expected. My current implementation of the cache (https://github.com/JuliaRobotics/RigidBodyDynamics.jl/compare/master...ferrolho:hf/wrenches-cache) looks very odd. What's the reason why I can't just do it like this?

```julia
struct WrenchesCache <: AbstractTypeDict
    keys::Vector{UInt}
    values::Vector{Dict{BodyID,Wrench{<:Any}}}
end

Base.show(io::IO, ::WrenchesCache) = print(io, "WrenchesCache(…)")

WrenchesCache() = WrenchesCache([], [])
@inline valuetype(::Type{WrenchesCache}, ::Type{T}) where {T} = Dict{BodyID,Wrench{T}}
@inline makevalue(c::WrenchesCache, ::Type{T}) where {T} = Dict{BodyID,Wrench{T}}()
```

> Yeah, you should be able to just use the `fixed_transform` function (or actually just `body_fixed_frame_definition`) to get the transform from whatever frame you defined the contact point in to the body's default body-fixed frame.

Thanks for the tip! I was unaware of `body_fixed_frame_definition`. Anyway, I ended up going with `fixed_transform`:

```julia
for (body, cp_inds) ∈ robot.feet_bodies_to_cp_inds
    body_id = BodyID(body)
    externalwrenches[body_id] = zero(Wrench{T}, root_frame(robot.mechanism))
    body_to_root = transform_to_root(state, body)
    for cp_ind ∈ cp_inds
        cp_body = robot.contact_points[cp_ind]
        foot_to_body = fixed_transform(robot.mechanism, cp_body.frame, default_frame(body))
        cp_root = transform(cp_body, body_to_root * foot_to_body)
        ind_λᵢⱼ = (1:3) .+ (cp_ind - 1) * 3
        force_root = FreeVector3D(root_frame(robot.mechanism), @view λᵢ[ind_λᵢⱼ])
        externalwrenches[body_id] += Wrench(cp_root, force_root)
    end
end
```

(The source of the transform `foot_to_body` is a foot which has been removed from the kinematic tree by the time that loop executes.)

> How are you measuring allocations?

I am measuring the allocations with `@btime`. Here is a reproducible example:

```
] add https://github.com/ferrolho/RigidBodyDynamics.jl#hf/wrenches-cache
```

```julia
# Workaround for https://github.com/JuliaRobotics/RigidBodyDynamics.jl/issues/500
using LinearAlgebra; BLAS.set_num_threads(1)

using BenchmarkTools
import RigidBodyDynamics as RBD

joint_types = [RBD.QuaternionFloating{Float64}; [RBD.Revolute{Float64} for i = 1:10]]
mechanism = RBD.rand_tree_mechanism(Float64, joint_types...)

statecache = RBD.StateCache(mechanism)
wrenchescache = RBD.WrenchesCache(mechanism)

contact_points = [
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body1")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body2")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body3")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body4")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body5")), 0.0, 0.0, 0.0),
]

feet_bodies_to_cp_inds = Dict(
    RBD.findbody(mechanism, "body1") => 1:1,
    RBD.findbody(mechanism, "body2") => 2:2,
    RBD.findbody(mechanism, "body3") => 3:3,
    RBD.findbody(mechanism, "body4") => 4:4,
    RBD.findbody(mechanism, "body5") => 5:5,
)

let
    function f(mechanism, statecache, wrenchescache, contact_points, feet_bodies_to_cp_inds, λᵢ::AbstractVector{T}) where {T}
        state = statecache[T]
        externalwrenches = wrenchescache[T].wrenches

        empty!(externalwrenches)

        for (body, cp_inds) ∈ feet_bodies_to_cp_inds
            body_id = RBD.BodyID(body)
            externalwrenches[body_id] = zero(RBD.Wrench{T}, RBD.root_frame(mechanism))
            body_to_root = RBD.transform_to_root(state, body)
            for cp_ind ∈ cp_inds
                cp_body = contact_points[cp_ind]
                cp_root = RBD.transform(cp_body, body_to_root)
                ind_λᵢⱼ = (1:3) .+ (cp_ind - 1) * 3
                force_root = RBD.FreeVector3D(RBD.root_frame(mechanism), @view λᵢ[ind_λᵢⱼ])
                externalwrenches[body_id] += RBD.Wrench(cp_root, force_root)
            end
        end
    end

    λᵢ = rand(15)  # 5 contacts * 3

    # f(mechanism, statecache, wrenchescache, contact_points, feet_bodies_to_cp_inds, λᵢ)

    # @code_warntype f(mechanism, statecache, wrenchescache, contact_points, feet_bodies_to_cp_inds, λᵢ)

    @btime $f($mechanism, $statecache, $wrenchescache, $contact_points, $feet_bodies_to_cp_inds, $λᵢ)
end
```

Which, on my machine, results in:

```
400.420 ns (5 allocations: 560 bytes)
```

tkoolen, 2021-03-16:
Thanks for the reproducer. Looks like the `@view` allocation elimination described in https://github.com/JuliaLang/julia/blob/master/HISTORY.md#compilerruntime-improvements-1 is still somewhat limited. Here's a fixed version:

```julia
using LinearAlgebra; BLAS.set_num_threads(1)

using StaticArrays
using BenchmarkTools
import RigidBodyDynamics
const RBD = RigidBodyDynamics

joint_types = [RBD.QuaternionFloating{Float64}; [RBD.Revolute{Float64} for i = 1:10]]
mechanism = RBD.rand_tree_mechanism(Float64, joint_types...)

statecache = RBD.StateCache(mechanism)
wrenchescache = RBD.WrenchesCache(mechanism)

contact_points = [
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body1")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body2")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body3")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body4")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body5")), 0.0, 0.0, 0.0),
]

feet_bodies_to_cp_inds = Dict(
    RBD.findbody(mechanism, "body1") => 1:1,
    RBD.findbody(mechanism, "body2") => 2:2,
    RBD.findbody(mechanism, "body3") => 3:3,
    RBD.findbody(mechanism, "body4") => 4:4,
    RBD.findbody(mechanism, "body5") => 5:5,
)

function f(mechanism, statecache, wrenchescache, contact_points, feet_bodies_to_cp_inds, λᵢ::AbstractVector{T}) where {T}
    state = statecache[T]
    externalwrenches = wrenchescache[T].wrenches

    empty!(externalwrenches)

    for (body, cp_inds) ∈ feet_bodies_to_cp_inds
        body_id = RBD.BodyID(body)
        externalwrenches[body_id] = zero(RBD.Wrench{T}, RBD.root_frame(mechanism))
        body_to_root = RBD.transform_to_root(state, body)
        for cp_ind ∈ cp_inds
            cp_body = contact_points[cp_ind]
            cp_root = RBD.transform(cp_body, body_to_root)
            ind_λᵢⱼ = (1:3) .+ (cp_ind - 1) * 3
            force_root = RBD.FreeVector3D(RBD.root_frame(mechanism), SVector{3}(@view λᵢ[ind_λᵢⱼ]))
            externalwrenches[body_id] += RBD.Wrench(cp_root, force_root)
        end
    end
end

λᵢ = rand(15)  # 5 contacts * 3
@btime f($mechanism, $statecache, $wrenchescache, $contact_points, $feet_bodies_to_cp_inds, $λᵢ)
```

Note the `SVector{3}`.

`FreeVector3D` is actually parameterized by the vector type, not just the scalar type (which is a little inconsistent with some of the other types; sorry about that). This came in handy in certain applications involving frame-annotated views of subvectors of decision variables. But since it adds another layer of indirection on top of the dynamically-sized view into a heap-allocated vector, the compiler probably gives up on trying to prove that the view is short-lived and creating a stack-allocated variable for the vector view that references the heap-allocated vector .

closed 2021-03-16 by tkoolen

tkoolen, 2021-03-16:
Here's another version that also fixes the `MyWrenches` hack:

```julia
using LinearAlgebra; BLAS.set_num_threads(1)

using StaticArrays
using BenchmarkTools
import RigidBodyDynamics
const RBD = RigidBodyDynamics

joint_types = [RBD.QuaternionFloating{Float64}; [RBD.Revolute{Float64} for i = 1:10]]
mechanism = RBD.rand_tree_mechanism(Float64, joint_types...)

statecache = RBD.StateCache(mechanism)

struct WrenchesCache{M} <: RBD.AbstractTypeDict
    mechanism::RBD.Mechanism{M}
    keys::Vector{UInt}
    values::Vector # Can also be `Any`
end

Base.show(io::IO, ::WrenchesCache{M}) where {M} = print(io, "WrenchesCache{$M}(…)")

WrenchesCache(mechanism::RBD.Mechanism{M}) where {M} = WrenchesCache{M}(mechanism, [], [])

@inline function RBD.valuetype(::Type{WrenchesCache{M}}, ::Type{T}) where {M, T}
    RBD.BodyDict{RBD.Wrench{T}}
end

@inline function RBD.makevalue(c::WrenchesCache, ::Type{T}) where {T}
    rootframe = RBD.root_frame(c.mechanism)
    RBD.BodyDict{RBD.Wrench{T}}(b => zero(RBD.Wrench{T}, rootframe) for b in RBD.bodies(c.mechanism))
end

wrenchescache = WrenchesCache(mechanism)

contact_points = [
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body1")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body2")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body3")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body4")), 0.0, 0.0, 0.0),
    RBD.Point3D(RBD.default_frame(RBD.findbody(mechanism, "body5")), 0.0, 0.0, 0.0),
]

feet_bodies_to_cp_inds = Dict(
    RBD.findbody(mechanism, "body1") => 1:1,
    RBD.findbody(mechanism, "body2") => 2:2,
    RBD.findbody(mechanism, "body3") => 3:3,
    RBD.findbody(mechanism, "body4") => 4:4,
    RBD.findbody(mechanism, "body5") => 5:5,
)

function f(mechanism, statecache, wrenchescache, contact_points, feet_bodies_to_cp_inds, λᵢ::AbstractVector{T}) where {T}
    state = statecache[T]
    externalwrenches = wrenchescache[T]

    # empty!(externalwrenches)

    for (body, cp_inds) ∈ feet_bodies_to_cp_inds
        body_id = RBD.BodyID(body)
        externalwrenches[body_id] = zero(RBD.Wrench{T}, RBD.root_frame(mechanism))
        body_to_root = RBD.transform_to_root(state, body)
        for cp_ind ∈ cp_inds
            cp_body = contact_points[cp_ind]
            cp_root = RBD.transform(cp_body, body_to_root)
            ind_λᵢⱼ = (1:3) .+ (cp_ind - 1) * 3
            force_root = RBD.FreeVector3D(RBD.root_frame(mechanism), SVector{3}(@view λᵢ[ind_λᵢⱼ]))
            externalwrenches[body_id] += RBD.Wrench(cp_root, force_root)
        end
    end
end

λᵢ = rand(15)  # 5 contacts * 3
@btime $f($mechanism, $statecache, $wrenchescache, $contact_points, $feet_bodies_to_cp_inds, $λᵢ)
```

That's about 170 ns on my machine.

ferrolho, 2021-03-16:
I can confirm that putting the `@view λᵢ[ind_λᵢⱼ]` inside an `SVector{3}` fixed the unexpected allocations. I am curious as to how you pinned down the culprit. Did you inspect the code with `@code_typed` (or something similar)?

As for the `WrenchesCache`, it is quite impressive how fast `BodyDict` is compared to the `Dict{BodyID,Wrench{<:Any}}` I was using before. On my machine, I went from `~270 ns` (already with the non-allocating version) to `~180 ns`! Do you think this `WrenchesCache` is something you would be interested in officially adding to RigidBodyDynamics.jl?

Finally, I believe it is extremely important for fields of `struct`s to have very well-defined types, right? Otherwise, the compiler will not be able to generate optimised code that specialises on the types of that `struct` (and will have to generate much more code to support all the types the fields of that `struct` could take). What I want to confirm is related to the non-concrete type of the `values` field in `WrenchesCache`. Why is it okay for it to be annotated only as a `Vector`? (Or even as `Any`?) Are we allowed to get away with that because you manually annotate the type of `values` upon its return in the `Base.getindex` method for `AbstractTypeDict`s?

tkoolen, 2021-03-17:
> I am curious as to how you pinned down the culprit. Did you inspect the code with `@code_typed` (or something similar)?

Type stability is not the issue, so `@code_typed` is not really that useful. First I narrowed down what was causing the allocation by commenting out pieces of code (basically verifying your finding). This pointed to the `@view` code, and since the non-allocating `view` optimization is kind of a new thing and I've had to work around it many times before, that's what I suspected was the issue.

> Do you think this `WrenchesCache` is something you would be interested in officially adding to RigidBodyDynamics.jl?

Yeah, sure. It's getting to be a bit of a zoo of cache types and it would be nice to generalize this stuff a bit more, but I suppose it's fine for now.

> Why is it okay for it to be annotated only as a `Vector`? (Or even as `Any`?) Are we allowed to get away with that because you manually annotate the type of `values` upon its return ([here](https://github.com/JuliaRobotics/RigidBodyDynamics.jl/blob/c86bc6eaee29eee00205475507e70845841f4f12/src/caches.jl#L10)) in the `Base.getindex` method for `AbstractTypeDict`s?

Yeah, It's basically https://docs.julialang.org/en/v1/manual/performance-tips/#Annotate-values-taken-from-untyped-locations. We're helping the compiler predict the value type given just the key type `T` passed into `getindex` (by implementing `valuetype`). This essentially means that we're vouching for the type stability (predictability of output _type_ given input _types_, not values) of the `getindex` function, even though `keys` and `values` could even be modified arbitrarily outside of `getindex`.

ferrolho, 2021-03-18:
Your answers are very clear. Thanks! And thank you very much for the huge help throughout this thread overall.