JuliaRobotics/RigidBodyDynamics.jl #630: Symbolic Derivatives with SymPy do not work

traikodinev, 2022-03-21:
As of the latest version, the example is broken.

https://github.com/JuliaRobotics/RigidBodyDynamics.jl/blob/master/examples/6.%20Symbolics%20using%20SymPy/6.%20Symbolics%20using%20SymPy.jl

Using version 2.2.0 works.

The issue is that Quaternion does not support type Sym. I can update later with more info.

ferrolho, 2024-11-08:
@blegat, I was exploring the source of _Dionysos.jl_ and came across [BipedRobot/src/piracy.jl](https://github.com/dionysos-dev/Dionysos.jl/blob/master/BipedRobot/src/piracy.jl). Is that somehow correlated to this issue? Was the type piracy a workaround? Thanks in advance and sorry for the unsolicited tag... :P

ferrolho, 2024-11-10:
The issue can be fixed with
```
# Type piracy needed in order to make Symbolics.jl types interact well with Quaternions.jl.
# This should be avoided — see https://docs.julialang.org/en/v1/manual/style-guide/#avoid-type-piracy.
function Base.:/(q::Quaternions.Quaternion, x::Symbolics.Num)
    return Quaternions.Quaternion(q.s / x.val, q.v1 / x.val, q.v2 / x.val, q.v3 / x.val)
end
```
which should be included with #648 when that gets merged.