JuliaRobotics/RigidBodyDynamics.jl #642: Package does not build on Julia v1.11

blegat, 2024-10-30:
On Julia v1.11, building RigidBodyDynamics fail:
```julia
pkg> build RigidBodyDynamics
    Building RigidBodyDynamics → `~/.julia/dev/RigidBodyDynamics/deps/build.log`
ERROR: Error building `RigidBodyDynamics`, showing the last 100 of log: 
    @ ./Base.jl:557 [inlined]
 [23] include_package_for_output(pkg::Base.PkgId, input::String, depot_path::Vector{String}, dl_load_path::Vector{String}, load_path::Vector{String}, concrete_deps::Vector{Pair{Base.PkgId, UInt128}}, source::Nothing)
    @ Base ./loading.jl:2790
 [24] top-level scope
    @ stdin:5
in expression starting at /home/blegat/.julia/packages/Literate/VQn4b/src/Literate.jl:1
in expression starting at stdin:5
           ✗ Literate
  0 dependencies successfully precompiled in 2 seconds. 15 already precompiled.

ERROR: LoadError: The following 1 direct dependency failed to precompile:

Literate 

Failed to precompile Literate [98b081ad-f1c9-55d3-8b20-4c87d4299306] to "/home/blegat/.julia/compiled/v1.11/Literate/jl_UyDaNl".
ERROR: LoadError: ArgumentError: Package REPL does not have StyledStrings in its dependencies:
- Note that the following manifests in the load path were resolved with a different
  julia version, which may be the cause of the error. Try to re-resolve them in the
  current version, or consider deleting them if that fails:
    /home/blegat/.julia/dev/RigidBodyDynamics/examples/Manifest.toml (v1.9.0)
- You may have a partially installed environment. Try `Pkg.instantiate()`
  to ensure all packages in the environment are installed.
- Or, if you have REPL checked out for development and have
  added StyledStrings as a dependency but haven't updated your primary
  environment's manifest file, try `Pkg.resolve()`.
- Otherwise you may need to report an issue with REPL
Stacktrace:
 [1] macro expansion
   @ ./loading.jl:2226 [inlined]
```
Unfortunately, this is done in CI by the `buildpkg` github action. The reason for the failure is that it is executing the `deps/build.jl` file which executes the notebooks. This checks out the environment in `examples/Manifest.toml` as this environment was set up for Julia v1.9, it creates compatibility issues.

ferrolho, 2024-11-08:
@blegat can you check if this is resolved with the latest release v2.5.0? Thanks!

blegat, 2024-11-08:
I confirm that it's fixed now thanks !

closed 2024-11-08 by blegat